pybricks/support #225: [Bug] Number of parameters reported on error message off by one

BertLindeman, 2021-01-20:
**Describe the bug**
Error on ```sensor.ambient(None)``` is:
`TypeError: function takes 1 positional arguments but 2 were given`
But the `ambient` method has no parameter documented.

**To reproduce**
Steps to reproduce the behavior:
1. Run the program (below) on code.pybricks.com with a Spike Color sensor on port-B
2. Look at the console
3. See the error in the last lines

**Expected behavior**
Report that no parameters are supported for this method

**Screenshots**
None

**source**
```

#

from pybricks.pupdevices import ColorSensor
from pybricks.parameters import Port
from pybricks.tools import wait

# Initialize the sensor.
sensor = ColorSensor(Port.B)

# Get the ambient light intensity.
ambient = sensor.ambient()
print("ambient:", ambient)

# No parameter documented for ambient method
ambient = sensor.ambient(None)
ambient = sensor.ambient(None,None)

```

dlech, 2021-01-20:
This is a MicroPython thing: https://docs.micropython.org/en/latest/genrst/core_language.html#error-messages-for-methods-may-display-unexpected-argument-counts

BertLindeman, 2021-01-20:
Rats, I tread into the same hole again..

Confusing though . . 

Sorry.

Closing "as documented already"

Thanks David

closed 2021-01-20 by BertLindeman