pybricks/support #228: [Feature]  Allow CTRL-A CTRL-C to copy the text of the code.pybricks "about box"

BertLindeman, 2021-01-24:
**Is your feature request related to a problem? Please describe.**
While trying to make a new bug report I want to add the environment in which the situation is happening.

**Describe the solution you'd like**
Allow `copy` of the relevant text in the "about box".
E. g. the top title with the version.

**Describe alternatives you've considered**
An alternative is to take a screenshot, find where it got saved, copy and paste it into the new issue.
That is also taking a lot more storage and does not add extra info in this case.

**Additional context**
If it were possible something similar to `ev3dev-sysinfo` would be neat.

Another key or click combination might also be OKE.

dlech, 2021-01-24:
I was thinking the same thing. It is easy enough to allow selecting the text so that it can be copied. Is there anywhere else this might be useful? Probably error messages.

BertLindeman, 2021-01-24:
Sure: love to have a "select-all" key on the console.
And: what is the rationale on `paste` on the console?

Select-all is possible to do by hand, but not so efficient.

**Partly related**:
If possible without risk and easy to add info about the environment to the about box?
like ev3dev-info:
. Browser used
. Operating system
I think you want to know on new issues.

But those can be **later** improvements.

BertLindeman, 2021-01-24:
Oh, yes, error messages would be great too.

dlech, 2021-01-24:
> And: what is the rationale on `paste` on the console?

It is nice to be able to paste when using the REPL. Is this what you are asking?

BertLindeman, 2021-01-24:
Ah right, did not think of REPL.
There paste is very useful

dlech, 2021-01-24:
Fixed in pybricks-code v1.0.0-beta.3

Ctrl-A doesn't work in the terminal because that triggers raw REPL mode in MicroPython.

BertLindeman, 2021-01-24:
> 
> 
> Fixed in pybricks-code v1.0.0-beta.3
> 
> Ctrl-A doesn't work in the terminal because that triggers raw REPL mode in MicroPython.

Thanks David.

Seen working on the about box and the console log.
Excactly what I wanted.
Works on beta3:
Copied from the about box
```
Pybricks Code v1.0.0-beta.3

MicroPython for LEGO® Powered Up smart hubs.

Copyright (c) 2020-2021 The Pybricks Authors

LEGO® is a trademark of the LEGO Group of companies which does not sponsor, authorize or endorse this site.
```

Closing so soon ;-)

closed 2021-01-24 by BertLindeman

dlech, 2021-01-24:
> Ctrl-A doesn't work in the terminal because that triggers raw REPL mode in MicroPython.

I forgot to mention... it is still possible to *select all* in the terminal via the right-click context menu.

BertLindeman, 2021-01-24:
> 
> > Ctrl-A doesn't work in the terminal because that triggers raw REPL mode in MicroPython.
> 
> I forgot to mention... it is still possible to _select all_ in the terminal via the right-click context menu.

Yes, And I forgot to mention too:
CTRL-A selects all in the source window
CTRL-A selects all in the about box (off course not the buttons)

Right-click "Select all" selects all in the console log