pybricks/support #229: [Bug] Link in code.pybricks Documentation frame is refused on Microsoft EDGE browser

BertLindeman, 2021-01-24:
**Describe the bug**
In MS EDGE a link in the Documentation frame is attempted to be loaded into this Documentation frame.
MS EDGE browser refuses this. (See the screenshots below). 
. A copy of that link opens fine in a new MS EDGE tab.
. On Google Chrome the link is automatically opened in a new tab, and no problem occurs.

**To reproduce**
Steps to reproduce the behavior:
1. On win10 open MS EDGE
2. Go to 'code.pybricks.com'
2. Click on 'Settings'
3. Enable `Documentation`
4. Close "Settings'
5. Scroll down to the bottom of the help text
6. Click on `support page` in 
```Got questions or issues? Please share your findings on our support page so we can make Pybricks even better.```
7. See the error (like in the screenshots)

**Expected behavior**
Only open Pybricks documentation web pages in the documentation frame. For all other pages I would expect a new tab to be opened.

**Environment**
Win10
MS EDGE Versie 87.0.664.75 (Officiële build) (64-bits)
Google Chrome Version 87.0.4280.141 (Official Build) (64-bit)

**Circumvention**
Copy the link you want to click and paste that link into a new tab.
Or use Google Chrome

**Screenshots**
[image]

[image]

[image]

[image]

dlech, 2021-01-24:
Related: https://github.com/pybricks/pybricks-api/issues/21

BertLindeman, 2021-01-24:
> 
> 
> Related: [pybricks/pybricks-api#21](https://github.com/pybricks/pybricks-api/issues/21)

Ah, Laurens already found it, but that issue has been closed, but not fixed yet.
So I did not research enough, sorry.
Is there another item that is planned to fix this (that I obviously missed) 
If so, then this issue can also be closed.

BertLindeman, 2021-01-26:
Since 2021-01-26 this error situation no longer occurs.

Could it be that [include offline copy of docs #273](https://github.com/pybricks/pybricks-code/pull/273)
produces a different link, thus loading in a new tab?

Enough guessing, closing . . . .

closed 2021-01-26 by BertLindeman

laurensvalk, 2021-01-26:
@BertLindeman, Thanks for verifying. See https://github.com/pybricks/pybricks-api/commit/adf6e03bc94d34723fb43f05df14cb2219640128

BertLindeman, 2021-01-26:
Thanks for the link, Laurens. 
Guessing now is over ;-) 

Tested the links in the settings panel 
and the support link on the first page of the doc panel.