pybricks/support #230: [Bug] PUPdevice on TechnicHub reports ID=8 when no device is attached

BertLindeman, 2021-01-24:
**Describe the bug**
On a TechnicHub PUPDevice(port) reports device.info as id = 8
See [pybricks technical-info: list of identifications](https://github.com/pybricks/technical-info/blob/master/assigned-numbers.md)
Part of it:
| ID (hex) | Description |
| --- | --- |
| 0 (0x00) | no device |
| 8 (0x08) | Powered Up Lights |

**To reproduce**
Steps to reproduce the behavior:
1. Go to code.pybricks.com
2. Connect an hub
3. load the program from below
4. See console log 

**Expected behavior**
Would have expected the hubs all to response the same or similar at least.
. Either return an OSError ENODEV
or
. return a device ID of '0' zero

**Priority**
LOW

**Screenshots**
None now

**Source used**
``` python
#
# Source: Z:\home\bert\py\pybricks\issue\issue_#230.py
# simple_discover_devices.py

from pybricks import version
# from pybricks.tools import wait
from pybricks.iodevices import PUPDevice  # was LUMPDevice
from pybricks.parameters import Port
print(version)
hw_type = version[0]

if hw_type == "inventorhub" or hw_type == "primehub":  # aliases == the same firmware
    from pybricks.hubs import InventorHub
    hub = InventorHub()
elif hw_type == "technichub":
    from pybricks.hubs import TechnicHub
    hub = TechnicHub()
elif hw_type == "movehub":
    from pybricks.hubs import MoveHub
    hub = MoveHub()
else:
    raise BaseException("Unknown hub " + hw_type)
print(hw_type, "loaded\n\tBattery voltage:", hub.battery.voltage(), "mV")

def what_device_on_port(xport):
    try:
        device = PUPDevice(xport)
    except OSError as e:
        print("OSError", e, "on:", xport)
    except Exception as e:  # noqa:  E722 do not use bare 'except'
        print("General ", e, "occurred on ", xport)
    else:
        print("\t\tOn %-7s device.ID=%2d %s" % (xport, device.info()['id'], device.info()))
    return

if not hw_type == 'movehub':
    what_device_on_port(Port.A)
    what_device_on_port(Port.B)
what_device_on_port(Port.C)
what_device_on_port(Port.D)
if hw_type == 'primehub':
    what_device_on_port(Port.E)
    what_device_on_port(Port.F)

```

**Console log**
```
----------
('technichub', '3.0.0b1', 'v3.0.0b1 on 2021-01-22')
technichub loaded
        Battery voltage: 8178 mV
                On Port.A  device.ID= 8 {'id': 8}
                On Port.B  device.ID= 8 {'id': 8}
                On Port.C  device.ID= 8 {'id': 8}
                On Port.D  device.ID= 8 {'id': 8}

----------
('primehub', '3.0.0a13', 'v3.0.0a13-13-g087a11f7 on 2021-01-19')
primehub loaded
        Battery voltage: 7389 mV
OSError [Errno 19] ENODEV: 
<removed the cable etc explanations>
 on: Port.A
OSError [Errno 19] ENODEV:  on: Port.B
OSError [Errno 19] ENODEV:  on: Port.C
OSError [Errno 19] ENODEV:  on: Port.D
OSError [Errno 19] ENODEV:  on: Port.E
OSError [Errno 19] ENODEV:  on: Port.F

----------
('movehub', '3.0.0b1', 'v3.0.0b1 on 2021-01-22')
movehub loaded
        Battery voltage: 7422 mV
OSError [Errno 19] ENODEV on: Port.C
OSError [Errno 19] ENODEV on: Port.D
```

dlech, 2021-01-24:
Does the same thing happen on other hubs or only the TechnicHub?

BertLindeman, 2021-01-24:
> 
> 
> Does the same thing happen on other hubs or only the TechnicHub?

David, I only have the TechnicHub, MoveHub, PrimeHub and InventorHub.
The results of those are above in the console log.

I do not **know** for the other hubs.

dlech, 2021-01-24:
Sorry, I was just looking at the title of the issue.

laurensvalk, 2021-01-24:
Thanks for reporting this and for taking the time to test this on all hubs.

Possibly related: I recently noticed that it was possible to initialize `DCMotor` when in fact a regular `Motor` was attached. I don't remember for sure, but it could have been the Technic Hub indeed.

BertLindeman, 2021-01-24:
Funny, may be related, but not on a technichub:
The example for DCMotor **does** work on the movehub.
This program:
```
from pybricks.pupdevices import DCMotor
from pybricks.parameters import Port
from pybricks.tools import wait

# Initialize a motor without rotation sensors on port A.
example_motor = DCMotor(Port.A)

# Make the motor go clockwise (forward) at 70% duty cycle ("70% power").
example_motor.dc(70)

# Wait for three seconds.
wait(3000)

# Make the motor go counterclockwise (backward) at 70% duty cycle.
example_motor.dc(-70)

# Wait for three seconds.
wait(3000)
```

But also on the TechnicHub...
And the motor on port A is correctly reported as id=75 the inventor medium motor.

laurensvalk, 2021-01-24:
So maybe it's a different issue then. Thanks!

dlech, 2021-02-02:
> ```python
>     raise BaseException("Unknown hub " + hw_type)
> ```

(off topic) Python tip: Always use `Exception` instead of `BaseException` or even better, a more specific exception like `RuntimeError`.

dlech, 2021-02-03:
Measurement shows that something is pulling down the ID2 pin on Technic Hub.

Top line is ID2, second line is ID1. Measurements taken with no sensor connected, showing device detection sequence repeating.

# Technic Hub

[image]

# Move Hub

[image]

closed 2021-02-03 by dlech

BertLindeman, 2021-02-03:
> ```python
> BaseException
> ```
> (off topic) Python tip: Always use Exception instead of BaseException or even better, a more specific exception like RuntimeError.

Thank you, David.
I really like these little nudges in the direction of writing better python.

(more on topic) I now have a "powered-up leds" device that should be reported as "8".
Will test it the next firmware distribution.

BertLindeman, 2021-02-03:
Found build artifact and this issue **is** fixed.

trimmed extra `ENODEV` messages

```
('technichub', '3.0.0b1', 'v3.0.0b1-15-ga4b616f7 on 2021-02-03')
technichub loaded
        Battery voltage: 7839 mV
On: Port.A OSError [Errno 19] ENODEV: 
On: Port.B OSError [Errno 19] ENODEV:
On: Port.C OSError [Errno 19] ENODEV: 
On: Port.D OSError [Errno 19] ENODEV: 
```

And with leds attached to port C and D:
```
                On Port.C  device.ID= 8 {'id': 8}
                On Port.D  device.ID= 8 {'id': 8}
```