pybricks/support #233: Using the buttons

ZPhilo, 2021-01-26:
**Describe the issue**
On prime/inventor hub, Button.CENTER is listed, but pressing it aborts the program

**Suggested improvement**
Link to a program snippet that allows usage of Button.CENTER in custom user interface.

dlech, 2021-01-26:
On Prime/Inventor hubs, the center button is treated the same as the green button on other hubs. So if you really want to use it, you can catch the `KeyboardInterrupt` to use the button for input, but you have to be careful to make sure there is still a way to stop your program.

We can make an example, but we don't really want to widely encourage using it.

ZPhilo, 2021-01-26:
Makes sense... still think that something needs to be done to clarify this in Button documentation.

laurensvalk, 2021-01-26:
We could enable something like this. What do you think @ZPhilo?

```python

hub = PrimeHub()

hub.buttons.set_keyboard_interrupt(False)

# Now you can read the Button.CENTER too like the others!

# But of course now you have to take care of stopping yourself, or use a long press to shut down entirely.

hub.buttons.set_keyboard_interrupt(True)

# Now it works as usual again.

```

Until now, we had something more complicated in mind (e.g. a double click) that stopped us from doing it so far, but this would be quite simple.

For now, this would mean that the stop button in Pybricks Code would not work either, but we're fixing that in other ways.

dlech, 2021-01-26:
On the one hand, I would rather not add anything to keep things super simple. On the other hand, it would be cool if it took a button (or `None`) as the argument, then you could use the Bluetooth button as a stop button instead if you wanted to.

ZPhilo, 2021-01-26:
The simple scheme as proposed by Laurens seems fine to me. I was indeed a bit worried by the potential problem of lack of stop, but if a long press allows to take back control anyway, it's fine!

laurensvalk, 2021-01-26:
We could put this functionality in ‘’pybricks.tools.enable_keyboard_interrupt()’’ to keep the button interface simple.

Or better yet, check which micropython module already has this function and enable that.

laurensvalk, 2021-01-26:
> Or better yet, check which micropython module already has this function and enable that.

Guess what, it is already enabled :smile: 

```python
from pybricks.tools import wait

from micropython import kbd_intr

kbd_intr(-1)

print("Can't cancel program in the next 5 seconds!")
wait(5000)

print("Now you can stop again")
kbd_intr(3)
wait(5000)

```

So we just need to make sure we return `Button.CENTER` as well.

laurensvalk, 2021-01-26:
@ZPhilo: I know you like the online GitHub interface, so if you want to, you can add `&pb_Button_CENTER_obj` [here](https://github.com/pybricks/pybricks-micropython/blob/bf3072b6f7dd87b60e50d7c2130ca3c393a5709f/pybricks/hubs/pb_type_primehub.c#L34).

[image]

laurensvalk, 2021-01-27:
@JanderII is probably interested in this as well. Would this suit your application?

ZPhilo, 2021-01-27:
> @ZPhilo: I know you like the online GitHub interface, so if you want to, you can add `&pb_Button_CENTER_obj` [here](https://github.com/pybricks/pybricks-micropython/blob/master/pybricks/hubs/pb_type_primehub.c#L35).

Done!
BTW, I just noticed an error in buttons program snippet: it uses Button.BT instead of Button.BLUETOOTH

laurensvalk, 2021-01-27:
Thank you! I think now you just need to go to [your branch](https://github.com/ZPhilo/pybricks-micropython/tree/patch-1) and click on the pull request button.

laurensvalk, 2021-01-27:
> BTW, I just noticed an error in buttons program snippet: it uses Button.BT instead of Button.BLUETOOTH

You're right, thanks! It would be great if you want to [to change that also!](https://github.com/pybricks/pybricks-api/blob/master/examples/pup/hub_primehub/button_main.py)

ZPhilo, 2021-01-27:
Done for both (hopefully right ;) )

laurensvalk, 2021-01-28:
This has been added so we can close it. Thanks @ZPhilo!

closed 2021-07-06 by dlech