pybricks/support #234: [Bug] PrimeHub: Pressing two buttons at the "same" time results in "ValueError: incompatible .mpy file"

BertLindeman, 2021-01-26:
**Describe the bug**
Running a small button sample on the primehub seems to keep running and shows
``` ValueError: incompatible .mpy file```
after **each press** of the center button.

**To reproduce**
Steps to reproduce the behavior:
1. Go to code.pybricks.com
2. Enter the program:
``` python
from pybricks.hubs import PrimeHub
from pybricks.pupdevices import Motor
from pybricks.parameters import Port, Stop
from pybricks.tools import wait
from pybricks.experimental import getchar

hub = PrimeHub()
print("hub initiated")

help(hub)

# wait for any button to be pressed
while not any(hub.buttons.pressed()):
    wait(10)
print("After button pressed", hub.buttons.pressed())
```
3. Run the program
4. Try to press the left or right button **and** the center button **at the same time**.
5. After that, press the center button again. 
Looks like another run is attempted and something got wrong.
6. See the console log:
```
----------
hub initiated
object <PrimeHub> is of type PrimeHub
  battery -- <module 'battery'>
  buttons -- <offset>
  display -- <offset>
  imu -- <offset>
  light -- <offset>
  speaker -- <offset>
After button pressed (Button.LEFT,)
----------
ValueError: incompatible .mpy file
----------
ValueError: incompatible .mpy file
----------
ValueError: incompatible .mpy file
```

**Expected behavior**
The center button should probably interrupt and stop the program.

**Screenshots**
None now

The program can probably be shorter, but THIS is the program I ran into this issue.

BertLindeman, 2021-01-26:
This could well be a red herring.
Pressing the center button while no user program is running on the primehub, also issues the error message.
```
ValueError: incompatible .mpy file
```

BertLindeman, 2021-01-26:
So my question may be:
What is the center button supposed to do?

Also now saw #233

laurensvalk, 2021-01-26:
> This could well be a red herring.
> Pressing the center button while no user program is running on the primehub, also issues the error message.
> 
> ```
> ValueError: incompatible .mpy file
> ```

This is it :arrow_up: 

It happens because there's currently no default script in there (like we have on TechnicHub). This should go away once we have implemented user file storage on this hub.

> What is the center button supposed to do?

Once we implement it, this: #139.

BertLindeman, 2021-01-26:
Thanks  Laurens.

So closing **this** issue as there is an item to fix this.

closed 2021-01-26 by BertLindeman