pybricks/support #235: Why do I get an error when trying to call hub.imu.angular_velocity() ?

xserk, 2021-01-28:
**Question**
Why do I get an error when trying to call hub.imu.angular_velocity() ?

**Context**
I'm trying to run a code found in the documentation using a technic hub but I get the error AttributeError: 'IMU' object has no attribute 'angular_velocity' 

Thank you

**Screenshots**
[image]

[image]

laurensvalk, 2021-01-28:
Which version are you running?

To find out, run:

```python

from pybricks import version
print(version)

```

You may need to update your firmware because these methods were added only recently.

xserk, 2021-01-28:
('technichub', '3.0.0a13', 'v3.0.0a13 on 2021-01-08')

laurensvalk, 2021-01-28:
Yeah, try updating the firmware in that case. :)

xserk, 2021-01-28:
I loaded the firmware via the button you have on the test page, should i drag and drop latest firmware there? Will I be able to push the play button to load the code after that? Thank you.

laurensvalk, 2021-01-28:
Hmm, the button should give you the latest.

I wonder if you might be running an older version of our app.

Can you try doing a hard refresh using `ctrl+F5`?

laurensvalk, 2021-01-28:
If you go to settings, and click `about`, you should see this version:

[image]

xserk, 2021-01-28:
That was it! Chrome had a cached version of the page. Thank you!

laurensvalk, 2021-01-28:
Cool!

> Will I be able to push the play button to load the code after that? Thank you.

With this option, you can:

[image]

dlech, 2021-01-29:
The app now has a *Check for Update* in case automatic updates are missed.

closed 2021-01-29 by dlech