pybricks/support #236: [Doc] Document supported MicroPython modules

laurensvalk, 2021-01-28:
**Describe the issue**
It is not clear which MicroPython modules are enabled.

**Suggested improvement**
Document which modules and functions are enabled. Link to the corresponding MicroPython docs, or document them partly in the Pybricks docs.

Then we can also include examples for things like #233.

See also #140.

dlech, 2021-01-28:
By preferring the "u" prefixed names, we can also include stubs in the `pybricks` PyPi package to get code completion for these without conflicting with standard Python libraries. (Technically, we will probably have to have multiple packages because of how `pip` works, e.g. `pybricks-urandom` but these can be a dependency of the main `pybricks` package.)

laurensvalk, 2021-01-28:
That would be nice.

If it is technically possible to keep those in the `pybricks-api` repo somehow, that would make things easy enough to maintain.

dlech, 2021-01-28:
Yes, Poetry allows multiple packages. It might be nice to move things under a `src/` directory at the same time so we don't end up with too many top-level folders.

```
src/
  micropython/
  pybricks/
  urandom/
```

dlech, 2023-02-04:
This is done.

closed 2023-02-04 by dlech