pybricks/support #237: [Bug] Primehub left and right button long pressed also report then BLUETOOTH button

BertLindeman, 2021-01-28:
**Describe the bug**
A simultaneous long press of both the left and right button also reports the Bluetooth button

**To reproduce**
Steps to reproduce the behavior:
1. on code.pybricks.com run the program shown below
2. Press and hold the left button
Press and hold the right button
3. See the buttons reported in the console log

```
wait for any button to be pressed
        The center button stops the program run
Button(s) pressed ()
Button(s) pressed (Button.RIGHT,)
Button(s) pressed (Button.LEFT, Button.RIGHT)
Button(s) pressed (Button.LEFT, Button.BLUETOOTH)
Button(s) pressed (Button.LEFT, Button.RIGHT)
Button(s) pressed ()
Button(s) pressed (Button.CENTER,)
Traceback (most recent call last):
  File "main.py", line 20, in <module>
KeyboardInterrupt: 
```

**Expected behavior**
Only the pressed buttons to be reported

**Screenshots**
none now

**the program used**
``` python
# source Z:\home\bert\py\pybricks\tools\show_keyPressed.py
from pybricks import version
from pybricks.hubs import PrimeHub
from pybricks.tools import wait

hub = PrimeHub()
print("hub initiated", version)
print("wait for any button to be pressed")
print("\tThe center button stops the program run")

was_key = None
while True:
    cur_key = hub.buttons.pressed()
    if cur_key == was_key:
        pass
    else:
        print("Button(s) pressed", cur_key)
        was_key = cur_key
    wait(10)  # do not overload the system
```

**DR;TL**
Test ran on a Spike primehub: ('primehub', '3.0.0b1', 'v3.0.0b1-6-g9fc79e68 on 2021-01-27')
Tested also on an Inventorhub ('primehub', '3.0.0b1', 'v3.0.0b1-1-gbf3072b6 on 2021-01-22')
**Same result**

Separately pressed, each button is reported correct.
Pressing [left] and [bluetooth] works ok
Pressing [right] and [bluetooth] works ok

The order in which the buttons are pressed has no influence.
So:
. press and hold [left]
. press and hold [right]
behaves the same as:
. press and hold [right]
. press and hold [left]

Looking at the buttons because of #233

BertLindeman, 2021-01-29:
Went back to older firmware and there this problem did **not** occur.
`('primehub', '3.0.0a13', 'v3.0.0a13-3-g7c9c2d68 on 2021-01-12')`

No special reason why I picked **this** firmware.

laurensvalk, 2021-01-29:
Thanks, nice find!

> Went back to older firmware and there this problem did not occur.

That firmware didn't return `Button.CENTER` yet, which was only added a few days ago.

BertLindeman, 2021-01-29:
> 
> 
> Thanks, nice find!
> 
> > Went back to older firmware and there this problem did not occur.
> 
> That firmware didn't return `Button.CENTER` yet, which was only added a few days ago.

In this case the center button is not involved. But that change **could** be involved? I don't know.

laurensvalk, 2021-01-29:
How long did you have to press it before that button showed up?

```
Button(s) pressed (Button.LEFT, Button.RIGHT)
Button(s) pressed (Button.LEFT, Button.BLUETOOTH)   # <---  I don't seem to see this on v3.0.0b1-6-g9fc79e68
Button(s) pressed (Button.LEFT, Button.RIGHT)
```

BertLindeman, 2021-01-29:
> 
> 
> How long did you have to press it before that button showed up?
> 
> ```
> Button(s) pressed (Button.LEFT, Button.RIGHT)
> Button(s) pressed (Button.LEFT, Button.BLUETOOTH)   # <---  I don't seem to see this on v3.0.0b1-6-g9fc79e68
> Button(s) pressed (Button.LEFT, Button.RIGHT)
> ```

Ran this on the inventor hub:
```
hub initiated ('primehub', '3.0.0b1', 'v3.0.0b1-1-gbf3072b6 on 2021-01-22')
wait for any button to be pressed
        The center button stops the program run
Button(s) pressed ()
Button(s) pressed (Button.RIGHT,)
Button(s) pressed (Button.LEFT, Button.RIGHT)
Button(s) pressed (Button.LEFT, Button.BLUETOOTH)
Button(s) pressed (Button.LEFT, Button.RIGHT)
Button(s) pressed (Button.LEFT,)
Button(s) pressed ()
```
Timed by head about 2 seconds.

Then took the Spike hub and that sprayed messages at once:
```
hub initiated ('primehub', '3.0.0b1', 'v3.0.0b1-1-gbf3072b6 on 2021-01-22')
wait for any button to be pressed
        The center button stops the program run
Button(s) pressed ()
Button(s) pressed (Button.RIGHT,)
Button(s) pressed (Button.LEFT, Button.BLUETOOTH)
Button(s) pressed (Button.LEFT, Button.RIGHT)
Button(s) pressed (Button.LEFT, Button.BLUETOOTH)
Button(s) pressed (Button.LEFT, Button.RIGHT)
```
Strange looks like the same firmware..

I do not know of a way to test if it is hardware related.

(not related: will test the dual boot script)

laurensvalk, 2021-01-29:
The buttons are read via an analog to digital converter (several buttons on the same ADC, each with its own resistance combination). It's possible we don't have the right thresholds or perhaps we need to make the signal less jumpy.

And this is why we appreciate input from you and other users. We hadn't seen this yet, but now we know about it :)

dlech, 2021-02-02:
Same program run with ADC values printed (looks like the issue also exists when all three buttons (L, R, BT)  are pressed):

```
----------
hub initiated ('primehub', '3.0.0b1', 'v3.0.0b1-11-gb6d8de24-dirty on 2021-02-01')
wait for any button to be pressed
        The center button stops the program run
Button(s) pressed ()
3972 4044
Button(s) pressed (Button.LEFT,)
3973 2634
Button(s) pressed ()
3967 4047
Button(s) pressed (Button.RIGHT,)
3977 3123
Button(s) pressed ()
3967 4044
Button(s) pressed (Button.LEFT,)
3968 2634
Button(s) pressed (Button.LEFT, Button.RIGHT)
3976 2215
Button(s) pressed (Button.LEFT, Button.BLUETOOTH)
3967 2219
Button(s) pressed (Button.LEFT, Button.RIGHT)
3974 2216
Button(s) pressed (Button.LEFT, Button.BLUETOOTH)
3967 2219
Button(s) pressed (Button.LEFT, Button.RIGHT)
3975 2216
Button(s) pressed (Button.LEFT, Button.BLUETOOTH)
3967 2219
Button(s) pressed (Button.LEFT, Button.RIGHT)
3973 2215
Button(s) pressed (Button.LEFT, Button.BLUETOOTH)
3967 2219
Button(s) pressed (Button.LEFT, Button.RIGHT)
3971 2215
Button(s) pressed (Button.LEFT, Button.BLUETOOTH)
3967 2219
Button(s) pressed (Button.LEFT, Button.RIGHT)
3968 2215
Button(s) pressed (Button.LEFT, Button.BLUETOOTH)
3967 2219
Button(s) pressed (Button.LEFT, Button.RIGHT)
3975 2216
Button(s) pressed (Button.LEFT, Button.BLUETOOTH)
3967 2219
Button(s) pressed (Button.LEFT, Button.RIGHT)
3967 2218
Button(s) pressed (Button.LEFT, Button.BLUETOOTH)
3967 2219
Button(s) pressed (Button.LEFT, Button.RIGHT)
3971 2216
Button(s) pressed (Button.LEFT,)
3976 2634
Button(s) pressed ()
3974 4046
Button(s) pressed (Button.BLUETOOTH,)
3973 3615
Button(s) pressed (Button.RIGHT, Button.BLUETOOTH)
3971 2866
Button(s) pressed (Button.BLUETOOTH,)
3969 3615
Button(s) pressed ()
3977 4046
Button(s) pressed (Button.LEFT,)
3972 2634
Button(s) pressed (Button.LEFT, Button.BLUETOOTH)
3973 2447
Button(s) pressed (Button.LEFT,)
3976 2634
Button(s) pressed ()
3972 4043
Button(s) pressed (Button.LEFT,)
3967 2638
Button(s) pressed (Button.LEFT, Button.RIGHT, Button.BLUETOOTH)
3973 2083
Button(s) pressed (Button.LEFT, Button.RIGHT)
3967 2086
Button(s) pressed (Button.LEFT, Button.RIGHT, Button.BLUETOOTH)
3973 2081
Button(s) pressed (Button.LEFT, Button.RIGHT)
3967 2086
Button(s) pressed (Button.LEFT, Button.RIGHT, Button.BLUETOOTH)
3973 2082
Button(s) pressed (Button.LEFT, Button.RIGHT)
3967 2086
Button(s) pressed (Button.LEFT, Button.RIGHT, Button.BLUETOOTH)
3973 2083
Button(s) pressed (Button.LEFT, Button.RIGHT)
3967 2086
Button(s) pressed (Button.LEFT, Button.RIGHT, Button.BLUETOOTH)
3973 2082
Button(s) pressed (Button.LEFT, Button.RIGHT)
3967 2086
Button(s) pressed (Button.LEFT, Button.RIGHT, Button.BLUETOOTH)
3977 2082
Button(s) pressed (Button.LEFT, Button.RIGHT)
3967 2086
Button(s) pressed (Button.LEFT, Button.RIGHT, Button.BLUETOOTH)
3977 2080
Button(s) pressed (Button.LEFT, Button.RIGHT)
3967 2086
Button(s) pressed (Button.LEFT, Button.RIGHT, Button.BLUETOOTH)
3973 2082
Button(s) pressed (Button.LEFT, Button.RIGHT)
3967 2086
Button(s) pressed (Button.LEFT, Button.RIGHT, Button.BLUETOOTH)
3973 2082
Button(s) pressed (Button.LEFT, Button.RIGHT)
3967 2085
Button(s) pressed (Button.LEFT, Button.RIGHT, Button.BLUETOOTH)
3971 2083
Button(s) pressed (Button.LEFT, Button.RIGHT)
3967 2086
Button(s) pressed (Button.LEFT, Button.RIGHT, Button.BLUETOOTH)
3971 2082
Button(s) pressed (Button.LEFT, Button.RIGHT)
3967 2086
Button(s) pressed (Button.LEFT, Button.RIGHT, Button.BLUETOOTH)
3973 2082
Button(s) pressed (Button.LEFT, Button.RIGHT)
3967 2086
Button(s) pressed (Button.LEFT, Button.RIGHT, Button.BLUETOOTH)
3968 2082
Button(s) pressed (Button.LEFT, Button.BLUETOOTH)
3977 2445
Button(s) pressed (Button.LEFT,)
3973 2634
Button(s) pressed ()
3974 4045
```

current thresholds are:

```C

    if (value > 3654) {
        // no buttons pressed
    } else if (value > 3155) {
        *pressed |= PBIO_BUTTON_RIGHT_UP; // Bluetooth
    } else if (value > 2885) {
        *pressed |= PBIO_BUTTON_RIGHT;
    } else if (value > 2645) {
        *pressed |= PBIO_BUTTON_RIGHT_UP; // Bluetooth
        *pressed |= PBIO_BUTTON_RIGHT;
    } else if (value > 2454) {
        *pressed |= PBIO_BUTTON_LEFT;
    } else if (value > 2218) {
        *pressed |= PBIO_BUTTON_RIGHT_UP; // Bluetooth
        *pressed |= PBIO_BUTTON_LEFT;
    } else if (value > 2084) {
        *pressed |= PBIO_BUTTON_RIGHT;
        *pressed |= PBIO_BUTTON_LEFT;
    } else if (value > 1800) {
        *pressed |= PBIO_BUTTON_RIGHT_UP; // Bluetooth
        *pressed |= PBIO_BUTTON_RIGHT;
        *pressed |= PBIO_BUTTON_LEFT;
    } else {
        // hardware failure?
        return PBIO_ERROR_IO;
    }
```

closed 2021-02-02 by dlech