pybricks/support #240: [Feature] Uninstall button for Pybricks Code

laurensvalk, 2021-01-29:
**Is your feature request related to a problem? Please describe.**
Installing is easy now, but removing is a bit harder. 

**Describe the solution you'd like**

If installed, replace install by uninstall button.

[image]

**Describe alternatives you've considered**
Finding out where the uninstall button is in Chrome. We could add an explainer, but an uninstall button where the install button was, seems more intuitive.

dlech, 2021-01-29:
I'm not aware of any APIs that will uninstall the app, so the best we can do is document where to find it in the Chromium UI.

dlech, 2021-01-29:
We also have to consider that it won't be the same on Android vs Desktop.

laurensvalk, 2021-07-13:
>  the best we can do is document where to find it in the Chromium UI.

This is done [here](https://pybricks.com/install/running-programs/#using-pybricks-offline), so we can close this

closed 2021-07-13 by laurensvalk