pybricks/support #245: [Bug] DriveBase does not stop when program stops

laurensvalk, 2021-01-31:
> Sometimes when I stop the program, drivebase motors keep running forever. I may eventually stop them with a long press on hub's green button, but sometimes really need to remove the batteries so that the program can be stopped.

_Originally posted by @TechnicBRICKs in https://github.com/pybricks/support/discussions/244_

laurensvalk, 2021-01-31:
@TechnicBRICKS: can you also reproduce this in an example program that does not use the IMU?

TechnicBRICKs, 2021-01-31:
Well, since you opened an issue, I tried to repeat it it more systematically.

This quite sporadic event!
If using the IMU I noticed again after 5 runs, then ~20 runs, then ~10 runs.

When the issue is triggered, the best way to stop the motors seems to be releasing the BT connection via the Pybricks Code environment and then pressing the hub green button one or more times until the motors stop.

After removing the IMU calls, I've tried it again a some dozen times and never catch the same issue. That's only what I can say...
Even if it is a real bug, I'd call it a minor one.

laurensvalk, 2021-01-31:
> After removing the IMU calls, I've tried it again a some dozen times and never catch the same issue. That's only what I can say...
Even if it is a real bug, I'd call it a minor one.

In that case, it's probably the same as #232, so I'll close this one and move it there.

closed 2021-01-31 by laurensvalk