pybricks/support #249: [Bug] No progress-bar (circle) on flash to movehub

BertLindeman, 2021-02-03:
**Describe the bug**
Flash movehub on code.pybricks.com (Beta 6), shows no "progress-bar" circle.
Dropped the firmware zip onto the "Install Pybricks firmware" button
The flash also does not finish - Hub keeps blinking red-green-blue.

**To reproduce**
Steps to reproduce the behavior:
1. On code.pybricks.com drop the movehub firmware zip onto the "Install Pybricks firmware" button
2. Connect to the bootloader - the movehub
3. See no progress, No error in the development tools, hub keeps blinking (gave it 10+ minutes)
4. After hard stopping the hub with a long press  - short press and connect again
5. Firmware did **not** change: ('movehub', '3.0.0b1', 'v3.0.0b1 on 2021-01-22')

**Expected behavior**
New firmware neatly updated with progress indication.

**Screenshots**
none

**Environment**
Win 10
MS EDGE Versie 88.0.705.56 (Officiële build) (64-bits)

```
Pybricks Code v1.0.0-beta.6
```
The new firmware is from [Build #873](https://github.com/pybricks/pybricks-micropython/actions/runs/532424736)
But this should not be the problem, I think.

Maybe related: #183 

Will retry and report back here

BertLindeman, 2021-02-03:
The previous attempt (above) was done after
- `restart` due to beta level
- doing various program runs and other actions

Now I did a CRTL-F5 reload first.
Now the firmware flash on the movehub succeeds including the progress indication.

After Bluetooth connect, I get:

```
GATT Server is disconnected. Cannot retrieve services. (Re)connect first with `device.gatt.connect`.
```

Doing the connect once more: all is fine.
Firmware reported as 
```('movehub', '3.0.0b1', 'v3.0.0b1-15-ga4b616f7 on 2021-02-03')```

Assume the (lack of) speed of the movehub causes some timeouts.

dlech, 2021-02-03:
> GATT Server is disconnected. Cannot retrieve services. (Re)connect first with `device.gatt.connect`.

Probably the same as #56 . If you are ever able to reproduce the problem it would be super helpful to capture the Bluetooth traffic when the problem occurs. See [https://bleak.readthedocs.io/en/latest/troubleshooting.html#windows-10](here) for instructions.

BertLindeman, 2021-02-03:
> Probably the same as #56 . If you are ever able to reproduce the problem it would be super helpful to capture the Bluetooth traffic when the problem occurs. See [...]

Will try a few times.
There might be something wrong with your link above.
I did swap the URL and the description myself a few times  :smile:
Assume you mean [here](https://bleak.readthedocs.io/en/latest/troubleshooting.html#windows-10)

dlech, 2021-02-03:
> Assume you mean [here](https://bleak.readthedocs.io/en/latest/troubleshooting.html#windows-10)

Yeah, I got it backwards. Fixed now. 😄

BertLindeman, 2021-02-04:
atempted 5 times with bluetooth capture on.
No luck.
The progress circle takes a few seconds to start 
even after the hub starts blinking R-G-B

May be I was not patient enough and misinterpreted the situation.

Closing.

We can reopen if needed.

closed 2021-02-04 by BertLindeman