pybricks/support #251: More functional LEDs

dbfin, 2021-02-05:
I am using ev3dev with pybricks-micropython for a project of mine, and have encountered the following problem.

As it is now, the EV3Brick.light has only 2 methods `on` and `off` that allow only 3 colors: green, red, and orange (which is more like vermilion, i.e. red-orange). This is very restrictive given that under the hood we have 2 independent pairs of red and green lights on each side accepting values from 0 to 255. For example, a very nice yellow color can be obtained by setting green = 255, and red = 24.

Moreover, besides what is currently allowed, which is only a steady light, ev3dev actually allows a number of nice effects that again can be run independently on both pairs of LEDs.

I tried to write to `/sys/class/leds/...` directly from Python to no avail. Apparently, after investigation, it seems that there is a timer running in the background that proactively changes whatever values I try to write with its own (for example, sometimes I can even see say yellow I am writing there for a fraction of a second that is getting then overwritten). 

I understand that there is default animation when a user program is running, but my values get overridden even after I call `light.on` or even `light.off` -- in the latter case 0 is getting constantly written to `brightness`. If this is correct, I do not understand why this is done this way -- once a user uses `light.on` or `light.off` there is no way to get back to that default animation anyway, so should not any animation mechanism be stopped in that case anyway? Constantly writing something consumes energy unnecessarily it seems, while writing just once the needed brightness and also trigger `none` has exactly the same effect -- the steady color. Again, this is just my assumption based on my investigation and what I saw. 

So, the question is whether it is possible to start a program without the underlying timer/animation at all, or at least be able to stop it and write your own values to `/sys/class/leds`. Thank you!

dlech, 2021-02-05:
I would expect that after calling `ev3.light.off()` that you should be able to use `/sys/class/leds/...` directly. Can you share the code that reproduces the problem?

dbfin, 2021-02-05:
Here is a reproducible minimal code I extracted:
``` python
#!/usr/bin/env pybricks-micropython

from pybricks.hubs import EV3Brick
from pybricks.tools import wait

from os.path import exists

ev3 = EV3Brick()

ev3.light.off()

path = "/sys/class/leds/led0:red:brick-status/brightness"

print(exists(path))
with open(path, "r") as f: print(f.readline())
with open(path, "w") as f: f.write("255")
with open(path, "r") as f: print(f.readline())
wait(100)
with open(path, "r") as f: print(f.readline())

wait(3000)
```
The output is 
```
True
0

255

0
```
That is the value written is there initially, but gets overwritten back to 0 during that `wait(100)`. And you can actually see it. It blinks red for a fraction of a second and then turns black (off) again. It remains off for the remaining 3 seconds.

BertLindeman, 2021-02-05:
Could it that the `trigger` on your EV3 is **not** ```NONE```?

You could add a print of the trigger to your program:
``` python

trigger = "/sys/class/leds/led0:red:brick-status/trigger"
print(exists(trigger))
with open(trigger, "r") as f: print(f.readline())
```

The current trigger is the value between the square brackets.

Like on my brick now it is none:
```
[none] rfkill-any kbd-scrolllock kbd-numlock kbd-capslock kbd-kanalock kbd-shiftlock kbd-altgrlock kbd-ctrllock kbd-altlock kbd-shiftllock kbd-shiftrlock kbd-ctrlllock kbd-ctrlrlock mmc0 timer heartbeat default-on transient lego-ev3-battery-charging-or-full lego-ev3-battery-charging lego-ev3-battery-full lego-ev3-battery-charging-blink-full-solid rfkill0 phy0rx phy0tx phy0assoc phy0radio rfkill1

```

Or you could write `none` to the trigger in your program?

Bert

dbfin, 2021-02-05:
@BertLindeman I actually checked, and it was `[none]`. Moreover, what I really wanted initially was to write `timer` as the trigger, and then modify `delay_on` and `delay_off`, which appear in the directory only when the trigger is `timer`, so I tried but neither the delay files appeared in the directory, nor was the trigger changed to `[timer]`, it stayed `[none]`.

But let me try again in that bare bone program. Hold on a second.

dbfin, 2021-02-05:
Here it is:
``` Python
#!/usr/bin/env pybricks-micropython

from pybricks.hubs import EV3Brick
from pybricks.tools import wait

from os.path import exists
import re

ev3 = EV3Brick()

ev3.light.off()
wait(300)

path = "/sys/class/leds/led0:red:brick-status/"
brightness = path + "brightness"
trigger = path + "trigger"

assert exists(path)
assert exists(brightness)
assert exists(trigger)

with open(brightness, "r") as f: print(f.readline().rstrip('\n\r'))
with open(brightness, "w") as f: f.write("255")
with open(brightness, "r") as f: print(f.readline().rstrip('\n\r'))
wait(100)
with open(brightness, "r") as f: print(f.readline().rstrip('\n\r'))

with open(trigger, "r") as f: print(re.findall(r'\[.*\]', f.readline()))
with open(brightness, "w") as f: f.write("255")
with open(trigger, "w") as f: f.write("timer")
with open(trigger, "r") as f: print(re.findall(r'\[.*\]', f.readline()))
wait(100)
with open(trigger, "r") as f: print(re.findall(r'\[.*\]', f.readline()))

wait(1000)
```
A typical result is:
```
0
255
0
['[none]']
['[timer]']
['[none]']
```
So, the trigger is also overwritten within those 100 milliseconds back to `none`. Visually, I now see two red blinks of the left LED.

Do you guys have the EV3 brick and the latest ev3dev with pybricks-micropython to try this code yourself? Just to make sure you can reproduce the problem?

dbfin, 2021-02-05:
Just to make sure in the case different versions exist.

The SD card is flushed with ev3-micropython-v2.0.0-sd-card-image.img. Then updated as the usual Debian system.

The python code is run using the `brickrun` command as follows: 
```
brickrun --directory="/home/robot/Test" "/home/robot/Test/main.py"
```

Versions: pybricks-micropython 2.0.0, brickrun 1.2.1.

Please let me know if you need any additional information.

BertLindeman, 2021-02-05:
Sorry this post has become a bit long....

To simplify a bit more I stripped all ev3 processing from your program:
``` python

from os.path import exists
import re

from time import sleep

path = "/sys/class/leds/led0:red:brick-status/"
brightness = path + "brightness"
trigger = path + "trigger"

assert exists(path)
assert exists(brightness)
assert exists(trigger)

with open(brightness, "r") as f: print(f.readline().rstrip('\n\r'))
with open(brightness, "w") as f: f.write("255")
with open(brightness, "r") as f: print(f.readline().rstrip('\n\r'))
sleep(0.1) # wait(100)
with open(brightness, "r") as f: print(f.readline().rstrip('\n\r'))

with open(trigger, "r") as f: print(re.findall(r'\[.*\]', f.readline()))
with open(brightness, "w") as f: f.write("255")
with open(trigger, "w") as f: f.write("timer")
with open(trigger, "r") as f: print(re.findall(r'\[.*\]', f.readline()))
sleep(0.1) # wait(100)
with open(trigger, "r") as f: print(re.findall(r'\[.*\]', f.readline()))

sleep(0.1) # wait(1000)
```

Have to add: there is a complication here, I have a crontab task running each 5 minutes to verify the power status and let it blink the leds orange or red depending...

Running as root straight from terminal via ssl.
But power is OK, so lets run this, result here is also:
```
0
255
0
['[none]']
['[timer]']
['[none]']
```

Running as robot:
```
brickrun -r -- pybricks-micropython /home/bert/py/issue_251_noev3.py
0
255
0
['[none]']
['[timer]']
['[none]']
16:08:43 robot@ev3dev:~$ brickrun -r -- pybricks-micropython /home/bert/py/issue_251_noev3.py
0
255
0
['[none]']
['[timer]']
['[none]']
```

But one step simpler still from robot in an ssh terminal (editted excess out):
``` bash
 echo 'timer' > /sys/class/leds/led0:red:brick-status/trigger ;
 sleep 1s ; cat  /sys/class/leds/led0:red:brick-status/trigger
[timer] 
```
And it stays `timer`.
So do we know for **shure** that the `timer` indeed **is written** in python and that the value we see comes from a cache?

dbfin, 2021-02-05:
I think we do know that Python does write the value at least to brightness, because I see the red LED blinks for a fraction of a second before it goes back to black. How else we can check it?

dbfin, 2021-02-05:
And yes, manually via ssh I modify all values no problem and see colors and effects.

BertLindeman, 2021-02-05:
I tried a `f.close()` but I see no difference due to that

dbfin, 2021-02-05:
Usually the block `with` ensures that the file is closed when the code exits the block, but it does not hurt to make sure indeed. I also tried `f.flush()` to ensure that the number is written to the file, but the result is the same.

I am inclined to think that this is not the problem though, because I do see the red light for a moment. 

Do you know if there is indeed a job running in the background on a timer that produces the default animation (green light with increasing and decreasing intensity). If so, what happens to it when you set a color? One thing that could be is that it is kept alive but now writes a fixed color instead of that flashing animation.

BertLindeman, 2021-02-05:
> Usually the block `with` ensures that the file is closed when the code exits the block, but it does not hurt to make sure indeed. I also tried `f.flush()` to ensure that the number is written to the [...]

This goes deeper than my knowledge, I do not know of s task like that.
But there is a difference between writing to the sysfs files in python or directly on the commandline.
Dare I ask @dlech or @laurensvalk ?

dlech, 2021-02-05:
I found the relevant code: https://github.com/pybricks/pybricks-micropython/blob/v2/lib/pbio/src/light.c#L63-L96. This function is called every 32 milliseconds.

In order to make it work as expected, the `case PBIO_LIGHT_PATTERN_NONE:` should have been `return` instead of `break`.

dbfin, 2021-02-05:
@dlech Nice! So I was right about this behavior. 

I still don't know why this function would be constantly called after either `light.on()` or `light.off()`, because both assume constant light afterwards, but to minimize possible damage from side effects, it should probably be left as it is, except indeed `break` should be `return` I agree.

dbfin, 2021-02-07:
@dlech I see that you are the maintainer of the `pybricks-micropython` package on archive.ev3dev.org, if I am not mistaken,  so how usually fixes like this get into the repository, and how long it usually takes?

dlech, 2021-02-07:
We actually haven't had any critical bugs in v2.0 that needed fixing that we know of, so we haven't done any updates for it yet. We don't plan on adding any new features to v2, so there will only be bug fix releases if needed. This change seems low risk for breaking things, so I wouldn't be opposed to including it, but it would be nice to have some other fixes as well to make it worth doing a release.

laurensvalk, 2021-02-07:
It would be nice if we could somehow gauge the general interest for continued EV3 support beyond the stable 2.0 release.

Most of the features requested by @dbfin already exist in our upcoming v3.0 release (which excludes EV3).

If there is enough interest, we could bring EV3 back up to speed for a v3.X release. This shouldn't even too much work in terms of coding since there haven't been many changes. The main challenge would be the messaging and distribution surrounding an update. (But this would be equally true for a 2.1 update with fixes.)

BertLindeman, 2021-02-07:
> It would be nice if we could somehow gauge the general interest for continued EV3 support beyond the stable 2.0 release.

EV3 update to v3.x after pybricks got to v3 would be really nice.
So there are no unnecessary differences (as far as feasible) compared to pybricks firmware.

Would the update be restricted to an update to pybricks-micropython and the lib?
The EV3 getting started would need to add an ```apt update && apt-get install pybricks-micropython```
I understand that it would be a bit adventurous as you want to keep the install as simple as possible.
Alternatively, a new sd-card image would need communication and consent from LEGO-education.

laurensvalk, 2021-02-07:
I don't think the official LEGO Education SD Card image would get an update. I think teachers would prefer stability versus a fun feature concerning lights or parity with other hubs.

Depending on ev3dev's plans, it may be possible to release an update through a future update of their SD Card image.

But as you can see, this creates two versions, so we're not doing this just yet unless there is real demand for it.

BertLindeman, 2021-02-07:
> I don't think the official LEGO Education SD Card image would get an update. I think teachers would prefer stability versus a fun feature concerning lights or parity with other hubs.

You would not want to adapt your classroom handouts each month (a bit exaggerated)
I understand. 

> But as you can see, this creates two versions, so we're not doing this just yet unless there is real demand for it.

Understood Laurens,  it is as it is . .  :smile:
But fun it is anyway.

Thank you.

dbfin, 2021-02-08:
I don't think either LEGO would want to have all the trouble with an update to a new version 3, especially given that it seems they have somewhat abandoned the EV3 branch in favor of simplified versions, from Boost to Robot Inventor. However, if the new version 3 would be 100% compatible with the previous version 2, they might be interested in updating it. 

However, the main question is not about the new version 3, but about an update to the current version. A bug fix update, including the solution to this issue, does not have to involve LEGO at all as I understand. Just a regular update in the ev3dev's repo, right?

dlech, 2023-04-28:
The code has changed significantly since https://github.com/pybricks/support/issues/251#issuecomment-774231367 so I don't think this should be a problem any more. Duplicate issue #67 is still open if there are any further comments.

closed 2023-04-28 by dlech