pybricks/support #253: [Feature] Color.NONE vs. None

laurensvalk, 2021-02-10:
**Is your feature request related to a problem? Please describe.**

[image]

Right now, the return value can be `None` or of type `Color`. And `None` is treated as a special case by `detectable_colors`.

This can be confusing. It also breaks things like printing `measured_color.h` to see the hue.

**Describe the solution you'd like**
It might make more sense to just define `Color.NONE = Color(0, 0, 0)` explicitly. This also makes the difference with `Color.BLACK` easier to explain.

The only issue is backwards compatibility with the 2.0 release for EV3.

dlech, 2021-02-10:
:+1:  from me

laurensvalk, 2021-04-09:
Implementation done. Need to verify docs and examples.

laurensvalk, 2021-04-10:
This is now done via https://github.com/pybricks/pybricks-api/commit/68f626f1ef5778ba981becb610d701454ddffb3e and surrounding commits.

closed 2021-04-10 by laurensvalk