pybricks/support #254: [Question] No such file or directory when using ev3.speaker.play_file

Whatwaq, 2021-02-12:
**Question**
I have just started using micropython, so I apologise in advance if this is a relatively simple problem to solve

I want to play a mp3 file I have in my downloads folder on my ev3 brick, so I used this code:

```
#!/usr/bin/env pybricks-micropython
from pybricks.hubs import EV3Brick
from pybricks.ev3devices import Motor
from pybricks.parameters import Port
from pybricks.robotics import DriveBase

# Initialize the EV3 Brick.
ev3 = EV3Brick()

ev3.speaker.play_file('/Users/MyUsername/Downloads/hello.mp3')
```

However, it gives. me this error: 

```
Starting: brickrun --directory="/home/robot/Hello_world" "/home/robot/Hello_world/main.py"
----------
Traceback (most recent call last):
  File "/home/robot/Hello_world/main.py", line 11, in <module>
RuntimeError: Playing file failed: /Users/MyUsername/Downloads/hello.mp3: No such file or directory

----------
Exited with error code 1.
```

I'm using a mac OS. Does their file system work in some special way or something? I copied the file name directly from Finder's Right click -> Copy "hello.mp3" as pathname, so I don't really know what to do

I changed my username in the directory into "MyUsername" for privacy reasons(It's my irl name) but the same error occurs regardless.

**Screenshots**
[image]

Any help would be appreciated. Thank you!

BertLindeman, 2021-02-12:
Could it be that you are confusing the MAC filename and the filename on the EV3?

Maybe try to copy the `hello.mp3` to `/home/robot/`
and adapt the python program to use `/home/robot/hello.mp3`

Might also be that I did not correctly interpret your issue :smile:

Bert

laurensvalk, 2021-02-12:
Place `hello.mp3` in your project folder on your computer. So in the `hello_world` folder.

It will get downloaded to the EV3 automatically when you run your project.

In your code, use:

```python
ev3.speaker.play_file('hello.mp3')
```

closed 2021-02-12 by dlech