pybricks/support #256: [Bug] Hub took too long to respond

laurensvalk, 2021-02-13:
**Describe the bug**
Cannot install Technic Hub firmware in Microsoft Edge.

It pairs just fine, and appears to erase the flash as usual. The light is off for a bit, then starts blinking red/green/blue. But Pybricks Code does not appear to pick it up after.

**Expected behavior**
Start installation after erasing flash.

**Screenshots**
[image]

dlech, 2021-02-13:
Can you capture the Bluetooth traffic when this happens? https://bleak.readthedocs.io/en/latest/troubleshooting.html#capture

laurensvalk, 2021-02-14:
Will do. I'm stuck on the decoding step though (This app can't run on your PC); will have to try it later.

dlech, 2021-02-14:
I can decode it if that helps.

laurensvalk, 2021-02-14:
It appears to have loaded something; there must be something trivial with the command line interface not working.

[image]

laurensvalk, 2021-02-14:
Here's the etl file if I've done it right. Not sure this type of file contains any sysinfo? Feel free to delete if so.

[bth_hci.etl.txt](https://github.com/pybricks/support/files/5978342/bth_hci.etl.txt)

dlech, 2021-04-03:
[image]

For some reason, it looks like Windows is hitting a 30 second timeout waiting for the write response. At this point, the device is disconnected. But as soon as the disconnect request is made, we get the write response and the notification that we were expecting. It's like they are stuck in the Bluetooth stack or something (would need to get a proper Bluetooth sniffer to see for sure).

dlech, 2021-04-03:
[bth_hci.etl.txt.zip](https://github.com/pybricks/support/files/6253285/bth_hci.etl.txt.zip)

Decode file for the record.

dlech, 2021-04-12:
Does this still happen after rebooting Windows?

laurensvalk, 2021-04-13:
Yes. Still occurs consistently in `Pybricks Code v1.0.0-beta.14` in Microsoft Edge.

laurensvalk, 2021-04-19:
Need to go through old builds of Pybricks Code to see if it worked before.

laurensvalk, 2021-05-12:
```
BluetoothDebug[16:58:31] OnConnectionStatusChanged() status=0

BluetoothDebug[16:58:26] OnConnectionStatusChanged() status=1

BluetoothDebug[16:58:26] OnConnectionStatusChanged() status=0

BluetoothDebug[16:58:26] OnGattDiscoveryComplete() success=0

BluetoothDebug[16:58:26] Failed to get GATT services.

BluetoothDebug[16:58:26] Getting GATT Results failed.

BluetoothDebug[16:58:26] StartGattDiscovery()

BluetoothDebug[16:58:26] Discovering GATT services anew

BluetoothDebug[16:58:26] OnGattServicesChanged()

BluetoothDebug[16:57:55] OnGattDiscoveryComplete() success=1

BluetoothDebug[16:57:55] StartGattDiscovery()

BluetoothDebug[16:57:55] Discovering GATT services anew

BluetoothDebug[16:57:55] OnGattServicesChanged()

BluetoothDebug[16:57:54] StartGattDiscovery()

BluetoothDebug[16:57:54] Discovering GATT services anew

BluetoothDebug[16:57:54] OnGattServicesChanged()

BluetoothDebug[16:57:54] StartGattDiscovery()

BluetoothDebug[16:57:54] Discovering GATT services anew

BluetoothDebug[16:57:54] OnGattServicesChanged()

BluetoothDebug[16:57:54] OnConnectionStatusChanged() status=1

BluetoothDebug[16:57:53] StartGattDiscovery()

BluetoothDebug[16:57:53] No advertisement flags found.

FIDODebug[16:57:43] webauthn.dll version 2

USBUser[16:57:43] USB device added: path=\\?\usb#vid_0a12&pid_0001#5&12c8f4c0&0&2#{a5dcbf10-6530-11d2-901f-00c04fb951ed} vendor=2578 "", product=1 "", serial="", driver="BTHUSB", guid=09e66f12-17e9-4ba0-88ac-05955865aa82

USBError[16:57:43] Failed to read descriptor from node connection: A device attached to the system is not functioning. (0x1F)

USBUser[16:57:43] USB device added: path=\\?\usb#vid_80ee&pid_0021#5&12c8f4c0&0&1#{a5dcbf10-6530-11d2-901f-00c04fb951ed} vendor=33006 "VirtualBox", product=33 "USB Tablet", serial="", driver="HidUsb", guid=6fa93132-6cd2-4433-9d68-944da08d159b

USBError[16:57:43] Failed to read descriptor from node connection: A device attached to the system is not functioning. (0x1F)

BluetoothEvent[16:57:35] BluetoothAPI: 00001C10002F0400
```

laurensvalk, 2021-05-12:
```
BluetoothDebug[2021/05/12 17:00:36.031250] bluetooth_device_winrt.cc:804 OnGattDiscoveryComplete() success=1

BluetoothDebug[2021/05/12 17:00:35.387774] bluetooth_device_winrt.cc:788 StartGattDiscovery()

BluetoothDebug[2021/05/12 17:00:35.387772] bluetooth_device_winrt.cc:776 Discovering GATT services anew

BluetoothDebug[2021/05/12 17:00:35.387755] bluetooth_device_winrt.cc:760 OnGattServicesChanged()

BluetoothDebug[2021/05/12 17:00:35.203580] bluetooth_device_winrt.cc:788 StartGattDiscovery()

BluetoothDebug[2021/05/12 17:00:35.203576] bluetooth_device_winrt.cc:776 Discovering GATT services anew

BluetoothDebug[2021/05/12 17:00:35.203554] bluetooth_device_winrt.cc:760 OnGattServicesChanged()

BluetoothDebug[2021/05/12 17:00:35.024604] bluetooth_device_winrt.cc:788 StartGattDiscovery()

BluetoothDebug[2021/05/12 17:00:35.024601] bluetooth_device_winrt.cc:776 Discovering GATT services anew

BluetoothDebug[2021/05/12 17:00:35.024590] bluetooth_device_winrt.cc:760 OnGattServicesChanged()

BluetoothDebug[2021/05/12 17:00:35.023874] bluetooth_device_winrt.cc:737 OnConnectionStatusChanged() status=1

BluetoothDebug[2021/05/12 17:00:33.513082] bluetooth_device_winrt.cc:788 StartGattDiscovery()

BluetoothDebug[2021/05/12 17:00:33.411533] bluetooth_adapter_winrt.cc:211 No advertisement flags found.
```

dlech, 2021-05-12:
Hmm... not much useful there. I was at least expecting to see a disconnect after 30 seconds like we did in the packet capture.

dlech, 2021-05-12:
Does flashing with pybricksdev work on the same hardware?

laurensvalk, 2021-05-12:
Same hardware yes and no --- I'm running Windows on a VM.

And indeed, it seems to go wrong there with Pybricksdev as well, so I could dig a bit deeper.

```
Connected successfully!
Erasing flash and starting update
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "C:\Users\laure\Desktop\pybricksdev\pybricksdev\cli\__init__.py", line 327, in main
    asyncio.run(subparsers.choices[args.tool].tool.run(args))
  File "C:\Program Files\Python38\lib\asyncio\runners.py", line 43, in run
    return loop.run_until_complete(main)
  File "C:\Program Files\Python38\lib\asyncio\base_events.py", line 616, in run_until_complete
    return future.result()
  File "C:\Users\laure\Desktop\pybricksdev\pybricksdev\cli\__init__.py", line 202, in run
    await updater.flash(firmware, metadata, args.delay/1000)
  File "C:\Users\laure\Desktop\pybricksdev\pybricksdev\flash.py", line 300, in flash
    response = await self.bootloader_request(self.ERASE_FLASH)
  File "C:\Users\laure\Desktop\pybricksdev\pybricksdev\flash.py", line 267, in bootloader_request
    await self.write(data, delay, request.write_with_response)
  File "C:\Users\laure\Desktop\pybricksdev\pybricksdev\ble.py", line 140, in write
    await self.client.write_gatt_char(
  File "C:\Users\laure\Desktop\pybricksdev\.venv\lib\site-packages\bleak\backends\dotnet\client.py", line 733, in write_gatt_char
    raise BleakError(
bleak.exc.BleakError: Could not write value bytearray(b'\x11') to characteristic 00001626-1212-efde-1623-785feabcd123: Unreachable
```

dlech, 2021-05-12:
Does it work if you change the erase command to write without response?

laurensvalk, 2021-05-12:
Also confirming that I see this in Pybricksdev on both City Hub and Technic Hub.

> Does it work if you change the erase command to write without response?

It's been a long time since I worked on this code. The followowing appears to make no difference:

[image]

laurensvalk, 2021-05-12:
Also, the light does turn off briefly, so the hub appears to receive the erase command.

laurensvalk, 2021-05-12:
Without response, it mostly still doesn't work, but I saw it work once...

laurensvalk, 2021-05-12:
Hmm, looks like write without response and without reply + waiting about 5 minutes does the trick in pybricksdev. Maybe there's something else we can send to wake it up sooner. Might not be able to investigate tonight though.

laurensvalk, 2021-05-12:
So indeed, this quick hack does the trick:

```diff

diff --git a/pybricksdev/flash.py b/pybricksdev/flash.py
index 8ee5e5f..da8ecd5 100644
--- a/pybricksdev/flash.py
+++ b/pybricksdev/flash.py
@@ -11,6 +11,8 @@ from typing import Dict, Tuple
 from tqdm import tqdm
 import typing
 import zipfile
+import asyncio
+

 from .ble import BLERequestsConnection
 from .compile import save_script, compile_file
@@ -203,6 +205,8 @@ class BootloaderRequest():
         self.reply_len = struct.calcsize(data_format)
         if request_reply:
             self.reply_len += 1
+        if command == 0x11:
+            self.reply_len = 0
         self.write_with_response = write_with_response

     def make_request(self, payload=None):
@@ -223,7 +227,7 @@ class BootloaderConnection(BLERequestsConnection):

     # Static BootloaderRequest instances for particular messages
     ERASE_FLASH = BootloaderRequest(
-        0x11, 'Erase', ['result'], '<B'
+        0x11, 'Erase', ['result'], '<B', False, False
     )
     PROGRAM_FLASH_BARE = BootloaderRequest(
         0x22, 'Flash', [], '', False, False
@@ -273,7 +277,6 @@ class BootloaderConnection(BLERequestsConnection):
             return request.parse_reply(reply)

     async def flash(self, firmware, metadata, delay):
-
         # Firmware information
         firmware_io = io.BytesIO(firmware)
         firmware_size = len(firmware)
@@ -299,6 +302,7 @@ class BootloaderConnection(BLERequestsConnection):
         self.logger.debug("Erasing flash.")
         response = await self.bootloader_request(self.ERASE_FLASH)
         self.logger.debug(response)
+        await asyncio.sleep(3)

         # Get the bootloader ready to accept the firmware
         self.logger.debug('Request begin update.')

```

dlech, 2021-05-12:
Does Move hub work?

Can you do a Bluetooth packet capture for all hubs with the hack in place?

laurensvalk, 2021-05-14:
> Does Move hub work?

Yes, it works in Pybricks Code.

laurensvalk, 2021-05-14:
> Can you do a Bluetooth packet capture for all hubs with the hack in place?

[bth_hci_pybricksdev_hack.zip](https://github.com/pybricks/support/files/6478053/bth_hci_pybricksdev_hack.zip)

Notes:
- For Technic Hub I included a log without hack (which fails: ` Could not write value bytearray(b'\x11')`). Uploaded here for comparison with Pybricks Code log from before
- For Technic Hub I included a log with hack (which succeeds)
- For City Hub I included a log with the same hack. This fails, but that seems to be mainly because I hacked how the reply is treated, so Pybricksdev doesn't understand `b'\x11\x00'`. This can probably be worked around in another way.

dlech, 2021-05-14:
Thanks for this. In bth_hci_technichub_pybricksdev_default, it looks like the bootloader is sending the wrong response to Find By Type Value Request. This probably confuses the Windows Bluetooth stack since it is getting responses to commands it didn't send and not getting responses to commands that it did send. I recall that this request was added in a newer version of Windows and caused problems and had to be [fixed](https://github.com/pybricks/pybricks-micropython/commit/74bcde39c8cb27df22480f71f25cf4afe575511b) in our firmware. However, we can't fix it in the bootloader.

[image]

dlech, 2021-05-14:
In bth_hci_technichub_pybricksdev_hack, we can see that the notification for the erase command is never received. 

Also the Find By Type Value Request is not made, so the connection completes successfully.

dlech, 2021-05-14:
On City hub, we already know that using write without response will break Linux since it sends a response anyway and this causes BlueZ to disconnect.

The fact that City hub is getting the response eventually tells me it is not broken in the same way that Technic hub is.

dlech, 2021-05-14:
Can you do more captures with write with response and continuing even if the erase response notification is not received after some time?

dlech, 2021-05-14:
Basically, I think the workaround will be: if the erase response is not received within 5 seconds, assume it is not going to be received ever and continue with the next command. If the erase was successful and we just didn't receive the response, then all is good. If the hub is not responding, the next command will fail anyway. And if the erase failed but the hub is still responding, then the checksum of the flashed firmware will be wrong and the user will have to flash again anyway.

Depending on further tests, we may also need to add logic so that City hub does write with response and other hubs do write without response on the erase command.

laurensvalk, 2021-05-15:
With the updated Pybricksdev, it works now on Technic Hub.

On City Hub, it does not.

```
'pybricksdev' '--debug' 'flash' 'cityhub-firmware.zip' 
2021-05-15 11:41:47,550: DEBUG: asyncio: Using proactor: IocpProactor
Creating firmware
Searching for LEGO Bootloader
2021-05-15 11:41:48,151: DEBUG: bleak.backends.dotnet.scanner: Received 90:84:2B:2E:B5:6C: Unknown.
2021-05-15 11:41:48,165: DEBUG: bleak.backends.dotnet.scanner: Received 90:84:2B:2E:B5:6C: LEGO Bootloader.
Found: 90:84:2B:2E:B5:6C: LEGO Bootloader
Connecting to 90:84:2B:2E:B5:6C: LEGO Bootloader
2021-05-15 11:41:48,169: DEBUG: bleak.backends.dotnet.client: Connecting to BLE device @ 90:84:2B:2E:B5:6C
2021-05-15 11:41:48,293: DEBUG: bleak.backends.dotnet.client: _ConnectionStatusChanged_Handler: 1
2021-05-15 11:41:48,295: DEBUG: bleak.backends.dotnet.client: Get Services...
2021-05-15 11:41:51,035: INFO: bleak.backends.dotnet.client: Services resolved for BleakClientDotNet (90:84:2B:2E:B5:6C)
Connected successfully!
Erasing flash and starting update
2021-05-15 11:41:51,266: DEBUG: pybricksdev.flash: Getting device info.
2021-05-15 11:41:51,267: DEBUG: pybricksdev.flash: Clear and prepare reply
2021-05-15 11:41:51,267: DEBUG: pybricksdev.flash: Make and write request
2021-05-15 11:41:51,267: DEBUG: pybricksdev.ble: TX CHUNK: bytearray(b'U'), with response
2021-05-15 11:41:51,507: DEBUG: bleak.backends.dotnet.client: Write Characteristic 00001626-1212-efde-1623-785feabcd123 : bytearray(b'U')
2021-05-15 11:41:51,508: DEBUG: pybricksdev.flash: Awaiting reply
2021-05-15 11:41:51,510: DEBUG: pybricksdev.ble: DATA bytearray(b'U\x00\x00\x00\x10\x00P\x00\x08\xff\xff\x03\x08A')
2021-05-15 11:41:51,511: DEBUG: pybricksdev.flash: Info(version=268435456, start_addr=134238208, end_addr=134479871, type_id=65)
2021-05-15 11:41:51,512: DEBUG: pybricksdev.flash: Erasing flash.
2021-05-15 11:41:51,512: DEBUG: pybricksdev.flash: Clear and prepare reply
2021-05-15 11:41:51,513: DEBUG: pybricksdev.flash: Make and write request
2021-05-15 11:41:51,513: DEBUG: pybricksdev.ble: TX CHUNK: bytearray(b'\x11'), with response
2021-05-15 11:42:21,625: DEBUG: bleak.backends.dotnet.client: _ConnectionStatusChanged_Handler: 0
2021-05-15 11:42:22,336: DEBUG: bleak.backends.dotnet.client: _ConnectionStatusChanged_Handler: 1

```
Hangs, then eventually raises:

```
Exception has occurred: BleakError
Could not write value bytearray(b'\x11') to characteristic 00001626-1212-efde-1623-785feabcd123: Unreachable
```

laurensvalk, 2021-05-15:
On City Hub, when changing the flash request to write *without response* it does proceed, and we're seeing the same as before, the reply `b'\x11\x00'` does come back eventually, but apparently only after the next write, so the responses fail to match:

```
2021-05-15 11:49:52,894: DEBUG: pybricksdev.flash: Erasing flash.
2021-05-15 11:49:52,894: DEBUG: pybricksdev.flash: Clear and prepare reply
2021-05-15 11:49:52,894: DEBUG: pybricksdev.flash: Make and write request
2021-05-15 11:49:52,895: DEBUG: pybricksdev.ble: TX CHUNK: bytearray(b'\x11'), without response
2021-05-15 11:49:52,899: DEBUG: bleak.backends.dotnet.client: Write Characteristic 00001626-1212-efde-1623-785feabcd123 : bytearray(b'\x11')
2021-05-15 11:49:52,899: DEBUG: pybricksdev.flash: Awaiting reply
2021-05-15 11:50:02,910: INFO: pybricksdev.flash: did not receive erase reply, continuing anyway
2021-05-15 11:50:02,910: DEBUG: pybricksdev.flash: Request begin update.
2021-05-15 11:50:02,910: DEBUG: pybricksdev.flash: Clear and prepare reply
2021-05-15 11:50:02,915: DEBUG: pybricksdev.flash: Make and write request
2021-05-15 11:50:02,916: DEBUG: pybricksdev.ble: TX CHUNK: bytearray(b'D\xccm\x02\x00'), with response
2021-05-15 11:50:02,984: DEBUG: pybricksdev.ble: DATA bytearray(b'\x11\x00')
2021-05-15 11:50:02,985: DEBUG: bleak.backends.dotnet.client: Write Characteristic 00001626-1212-efde-1623-785feabcd123 : bytearray(b'D\xccm\x02\x00')
2021-05-15 11:50:02,986: DEBUG: pybricksdev.flash: Awaiting reply
```
Then fails with
```
Exception has occurred: ValueError
Unknown message: bytearray(b'\x11\x00')
```
Could we send a bogus message (get checksum?) just to clear up any pending replies before starting to send the firmware?

laurensvalk, 2021-05-15:
> Could we send a bogus message (get checksum?) just to clear up any pending replies before starting to send the firmware?

Yes, that works. I've pushed a [patch](https://github.com/pybricks/pybricksdev/commit/241c534514730af6f987c6bd96fa8491f03fe004) here.

So the flash erase command is now the same again for all hubs; this just moves the workaround instead of adding another one, so hopefully this isn't too bad.

laurensvalk, 2021-05-15:
Also, looking at the [log above](https://github.com/pybricks/support/issues/256#issuecomment-841632329) it's almost as if the hub does send that reply, since the script gets it back *before* the backend send the next one. But just the act of writing to the stack appears to unload that data.

dlech, 2021-05-15:
Did you try this patch on Linux? I'm pretty sure that it will cause BlueZ to disconnect (it did in the past). If that is the case, we need to do write without response (and the checksum hack) only on Windows.

Does it work on Windows if we just add an async sleep before sending the next message? I wonder if there is some other way we could "poke" the Windows Bluetooth stack to get it to see the notification or if the message really is getting held back on the hub.

dlech, 2021-05-15:
Could you also do a packet capture of City Hub on Windows with "write with response".

dlech, 2021-05-15:
Also, what Bluetooth adapter are you using?

laurensvalk, 2021-05-15:
Patch has only been tested on windows so far, hence pushed to a windows branch for now.

I also tried a plain async sleep (and a longer timeout, 10s), but this didn't seem to work. The old/bad response always came when attempting the next write.

I'll get back with captures and info on the adapter when I'm back at the PC.

laurensvalk, 2021-05-15:
I think I have this one: https://www.trust.com/en/product/18187-manga-bluetooth-4-0-adapter

laurensvalk, 2021-05-15:
> Could you also do a packet capture of City Hub on Windows with "write with response".

With the current Pybricksdev master:

[bth_hci_6b0e63bd_cityhub.zip](https://github.com/pybricks/support/files/6488162/bth_hci_6b0e63bd_cityhub.zip)

laurensvalk, 2021-05-15:
> Did you try this patch on Linux? I'm pretty sure that it will cause BlueZ to disconnect (it did in the past). If that is the case, we need to do write without response (and the checksum hack) only on Windows.

Indeed:

```
Exception has occurred: BleakError
Not connected

```

dlech, 2021-05-15:
> I think I have this one: https://www.trust.com/en/product/18187-manga-bluetooth-4-0-adapter

USB VID:PID?

dlech, 2021-05-15:
Decoded captures from https://github.com/pybricks/support/issues/256#issuecomment-841144528

[bth_hci_pybricksdev_hack (2).zip](https://github.com/pybricks/support/files/6488202/bth_hci_pybricksdev_hack.2.zip)

Decoded capture from https://github.com/pybricks/support/issues/256#issuecomment-841720523

[Uploading bth_hci_6b0e63bd_cityhub (2).zip…]()

dlech, 2021-05-15:
Decoded captures from https://github.com/pybricks/support/issues/256#issuecomment-841144528

[bth_hci_pybricksdev_hack (2).zip](https://github.com/pybricks/support/files/6488202/bth_hci_pybricksdev_hack.2.zip)

Decoded capture from https://github.com/pybricks/support/issues/256#issuecomment-841720523

[bth_hci_6b0e63bd_cityhub (2).zip](https://github.com/pybricks/support/files/6488207/bth_hci_6b0e63bd_cityhub.2.zip)

dlech, 2021-05-15:
> With the current Pybricksdev master:
> 
> [bth_hci_6b0e63bd_cityhub.zip](https://github.com/pybricks/support/files/6488162/bth_hci_6b0e63bd_cityhub.zip)

What happened when this ran? It looks like it got cut off before it ever to the point of sending LWP3 commands (probably due to other misbehavior due to buggy implementation of Bluetooth in the bootloader).

dlech, 2021-05-15:
I pushed a branch with a different workaround that shouldn't break Linux.

laurensvalk, 2021-05-16:
That patch (https://github.com/pybricks/pybricksdev/commit/c227f02a0c3af7d08e13ad29ddc44b73c1a0b107) works too :+1: 

So does this not break on other Windows machines, given that those were working before the patch?

dlech, 2021-05-16:
> So does this not break on other Windows machines, given that those were working before the patch?

It seems to be working fine on my Windows laptop.

dlech, 2021-05-16:
> `Bus 001 Device 011: ID 0a12:0001 Cambridge Silicon Radio, Ltd Bluetooth Dongle (HCI mode)`

I have a dongle with the same VID:PID. My Windows laptop doesn't seem to be able to load the driver for it, but I was able to use it in a Windows VM without problem (driver worked and pybricksdev worked without hacks).

What is the driver info for this dongle in your Windows VM?

dlech, 2021-05-16:
After disabling the built-in adapter, I was able to get this dongle to work on my laptop as well.

dlech, 2021-05-16:
I have pushed a branch to `pybricks-code` with similar fixes for testing.

laurensvalk, 2021-05-17:
I can confirm that https://github.com/pybricks/pybricks-code/commit/956b6f3c9664f1fbf547ddbee7c79160282d7e09 is working for me :tada:

dlech, 2021-05-17:
Hopefully you aren't the only person in the world that had this strange problem. :wink:

laurensvalk, 2021-05-17:
If we merge it, we'll never know :smile:

dlech, 2021-08-13:
Is there anything left to do on this issue?

laurensvalk, 2021-08-13:
Reading the messages above, I guess not.

closed 2021-08-13 by laurensvalk