pybricks/support #258: [Feature] pollable timer

dlech, 2021-02-20:
- We just enabled the `uselect` module on most hubs, so now it is possible to make an event loop using `poll()`. 
- In robotics, it is common to have a control loop that runs at a fixed time interval.

Putting these together, it would be useful to implement a pollable timer object. On ev3dev, this would be backed by a [timerfd](https://man7.org/linux/man-pages/man2/timerfd_create.2.html). On STM32 hubs, we would need to make a custom implementation based on the MicroPython stream object, but this is fairly trivial.

This object could also be used as a timer without using `poll()` as well.

```python
class Timer:
    """Pollable timer."""

    def set_time(self, interval: int, value: int) -> None:
        """Arms or disarms the timer.

        Args:
            interval: the time interval in milliseconds or 0 to only time out once
            value: the initial timeout in milliseconds

        If both ``interval`` and ``value`` are 0, the timer is disarmed.
        """

    def get_time(self) -> int, int:
        """Gets the time remaining and the interval

        Returns:
            Tuple containing the the time remaining in milliseconds and the interval in milliseconds.
            If both values are 0, the timer is disarmed. If ``interval`` is 0, the timer will only expire once.
        """

    def read(self) -> int:
        """Gets the number of times the timer has expired since the last ``read()``.
        """
```

Example:

```python
from pybricks.tools import Timer
from uselect import poll
from usys import stdin

timer = Timer()

p = poll()
p.register(timer)
p.register(stdin)

def control_loop(n):
   ...

def input_handler(c):
    ...

timer.set_time(0, 10)
while True:
    for obj, _ in p.poll():
        if obj == timer:
            control_loop(t.read())
        elif obj == stdin:
            input_handler(ord(stdin.read(1)))
```

dlech, 2021-02-20:
`poll()` already includes a timeout, so we can already do something like this:

```python
from uselect import poll
from usys import stdin

p = poll()
p.register(stdin)

def control_loop(n):
   ...

def input_handler(c):
    ...

timer.set_time(0, 10)
while True:
    for obj, _ in p.poll(10):
        if obj == stdin:
            input_handler(ord(stdin.read(1)))
        control_loop(t.read())
```

But this is not quite as robust and would introduce jitter in the control loop.

laurensvalk, 2021-02-20:
> This object could also be used as a timer without using poll() as well.

Before we add something like it to `tools`, it would be interesting to consider how it complements `StopWatch`, so we can make it clear to the user which one to use, and when. Perhaps we could call it `PollTimer`.

dlech, 2021-02-20:
The major difference is that `StopWatch` measures time while a `PollTimer` or `EventTimer` triggers events based on time.

Novakasa, 2021-04-15:
It would be nice if the timer had a `set_callback` feature, maybe as a argument to `set_time`

dlech, 2021-04-15:
> It would be nice if the timer had a callback feature, to call a function when the timer expires. Maybe as a `callback=None` argument to `set_time`

A pollable timer is actually an alternative to this so that we don't have to enable the MicroPython scheduler. The scheduler could "inject" the callback anywhere in a program which could have strange effects on the timing of the rest of the program. A pollable timer should be a bit more deterministic.

dlech, 2021-04-15:
On the other hand, enabling the scheduler would be trivial, so it is not out of consideration.

Novakasa, 2021-04-15:
My usecase (besides having a control loop) would be to use the timer at various places like as part of multiple instances (track switches on a train layout). It would be a bit awkward to manage all the timers centrally in `p.poll()` in the main loop. But, if it's non trivial on the implementation side, it is easy to work around with a Timer implementation that uses a `StopWatch` internally (and it needs to be updated in the control loop)

laurensvalk, 2024-10-02:
Do we still want this now that we have async support?

dlech, 2024-10-02:
No, I don't think we need this.

closed 2024-10-02 by dlech