pybricks/support #261: Feature request: Configure REPL echo over BT on Primehub

antonvh, 2021-02-22:
With the Primehub, on the LEGO firmware, uart and REPL on the Bluetooth virtual com port are always echoed to the USB virtual com port. And vice versa. That is annoying.

Can you make this configurable on Pybricks? Default REPL on USB and optional REPL on BT? 

Having a free uart/vcp on bt would open a lot of possibilities for inter-brick communication and brick-to-computer. You could then develop it via USB.

laurensvalk, 2021-02-22:
We might be doing things quite differently in Pybricks; a lot of it is still open and to be decided. USB and Bluetooth Classic are not enabled yet. We focus on BLE because all hubs have it, and because we can use it in [Pybricks Code](https://code.pybricks.com/).

At the moment, we are splitting the download-and-run communication stuff to a separate BLE service. This paves the way for easy hub-to-hub communication within user programs. (Without having to fall back to USB just to free up BT somehow.)

antonvh, 2021-02-22:
That’s great news! Even more flexibility. Looking forward to it!

I thought I’d send in this feature request early while 'things are still open and to be decided’.

A Virtual Com Port over BT Classic would allow for wireless SPIKE-Prime  <-> EV3 comms. Both ways. :P

laurensvalk, 2021-02-26:
We've summarized some of our plans for further discussion in #262, so we can close this one.

closed 2021-02-26 by laurensvalk