pybricks/support #267: [Question] Pybricksdev Python version issues

laurensvalk, 2021-03-01:
>I got pybricks installed on my Robot Inventor and was able to download and run a "hello world" type program using your chrome-based editor on windows 10.  But my main development is on linux so I wanted to get that running.  Preferably using the CLI tools in pybricksdev with the Visual Studio Code integration.
>
>When I run your suggested example 
>> poetry run pybricksdev run ble "Pybricks Hub" demo/shortdemo.py
>
>I get the following output and then it hangs forever -- even after my hub powers off.
>> Searching for Pybricks Hub
>> Connecting to 38:0B:3C:AA:B6:CE: Pybricks Hub
>
>I am using Python 3.9.  I thought I'd seen some indication that the code is being developed using 3.8 (though I can't find that right now).  Is the python version of issue?
>
>[Update]
>I deleted and re-cloned pybricksdev and now I found the reference to python 3.8.  Here's what the above poetry command now says:
>> The currently activated Python version 3.9.1 is not supported by the project (~3.8).
>> Trying to find and use a compatible version. 
>> Using python3.8 (3.8.7)
>> Creating virtualenv pybricksdev in /home/steve/LegoMindstorms/code/pybricksdev/.venv
>>
>>  RuntimeError
>>
>>  failed to find interpreter for Builtin discover of python_spec='python3.8'
>>
>>  at /usr/lib/python3/dist-packages/virtualenv/run/__init__.py:72 in build_parser
>

_Originally posted by @smr99 in https://github.com/pybricks/support/issues/167#issuecomment-787531984_

laurensvalk, 2021-03-01:
> But my main development is on linux so I wanted to get that running.

The Pybricks Code web editor also works on Linux in Chromium (or Chrome).

And when you click to install as an app, it's pretty nicely integrated too:

[image]

> Preferably using the CLI tools in pybricksdev with the Visual Studio Code integration.

Yes, we require 3.8 due to [one of the dependencies](https://github.com/pybricks/pybricksdev/issues/11).

You could install `pyenv` and install `3.8.X` with it for local use without changing your system Python version.

Pybricksdev is mainly intended as a convenience CLI tool for the core developers, so it's not very polished or user friendly (especially installation). If the community would like to see it further developped, we're hoping they'll contribute :)

smr99, 2021-03-04:
> > But my main development is on linux so I wanted to get that running.
> 
> The Pybricks Code web editor also works on Linux in Chromium (or Chrome).

That's fair.  It didn't work out of the box for me, but then I discovered the magic to turn on bluetooth within chrome (The chrome://flags/#enable-experimental-web-platform-features flag must be enabled.)  I can pair the hub now, and download a simple program that prints hello.

> > Preferably using the CLI tools in pybricksdev with the Visual Studio Code integration.
> 
> Yes, we require 3.8 due to [one of the dependencies](https://github.com/pybricks/pybricksdev/issues/11).

OK.  I have managed to get a python 3.8.6 environment, installed.  I kind of picked the version arbitrarily -- is there a recommended 'x' for 3.8.x?

Now the pybrickdev "run ble" command no longer hangs!  But it says it cannot find the pybricks hub in 5 seconds.  Even after I have paired it with the chrome code editor.  So clearly I have more debugging to do.   [I will note that I am able to connect to the hub in normal Lego Inventor mode using the "spike prime" CLI tools.]

laurensvalk, 2021-03-09:
Which version of the firmware are you using?

If you're not sure, you can run this:

```python
from pybricks import version

print(version)
```

closed 2021-05-15 by laurensvalk