pybricks/support #268: Hub at CI-build #925 enters IDLE, but no IDLE if an extra newline is printed [Bug]

BertLindeman, 2021-03-05:
**Describe the bug**
Running TechnicHub at firmware 
```('technichub', '3.0.0b2', 'v3.0.0b2-26-g79c88e10 on 2021-03-03')```

On the following program, the hub enters into IDLE
```
from pybricks import version
print(version)
```
The result looks like this:
```
('technichub', '3.0.0b2', 'v3.0.0b2-26-g79c88e10 on 2021-03-03')
>>>> IDLE
```

And my confusion rises high :confused: if I add a newline at the end of the print command.
Then IDLE is **not** entered. 

**To reproduce**
Steps to reproduce the behavior:
1. Flash the firmware from [build-925 on the TechnicHub](https://github.com/pybricks/pybricks-micropython/actions/runs/618719325)
2. Enter a print command as a program **without an extra newline** in it
3. Run the program
4. See IDLE being entered
5. add **```"\n"```** to the print command
6. Run the program
7. See NO IDLE being entered

_**Maybe related**
I am used to seeing ```----------``` as a separator between program runs.
Sometimes I see the printed string as a separator. 
But Cannot reproduce so forget this remark for now._

**Expected behavior**
No IDLE if I do not ask for it

**Screenshots**
- Simple program **with** newline ran a few times:

[image]

- Simple program **without** newline ran a few times (and needed to stop IDLE in between):

[image]
Note that IDLE is still active on this screenshot

BertLindeman, 2021-03-05:
The MoveHub does not support IDLE so it is amusing to see this:
[image]

First ran the program and "stop"
and then pressed the IDLE button.

It only looks like we are in IDLE: no typing shows up. 
This also goes for the TechnicHub.

Is it so that only the string ```>>>> IDLE``` is printed and no IDLE mode is really started?
But the "stop" button is activated.

**extra info**
- Win 10
- Pybricks Code v1.0.0-beta.7

laurensvalk, 2021-03-05:
``>>> IDLE`` looks like our internal program-upload utility shining through.

This is being actively revised right now as part of step 1 in #218.

You may want to stick to full beta releases for now. CI builds might be quite broken for a while.

BertLindeman, 2021-03-05:
Oke, thanks Laurens.
Closing as irrelevant at the moment.

closed 2021-03-05 by BertLindeman