pybricks/support #272: enable buttons class on single-button hubs

dlech, 2021-03-16:
See https://github.com/pybricks/support/issues/233

If we can disable the `KeyboardInterrupt` then it makes sense to be able to use the button in the user program.

laurensvalk, 2021-03-16:
Moving this issue to support since this is still being discussed. Issues in [pybricks-micropython](https://github.com/pybricks/pybricks-micropython/issues) are only for clear-cut issues that are ready to be implemented or fixed.

laurensvalk, 2021-03-16:
>> enable buttons class
>
> One consideration is that there is only one button, so using the same class might be a bit overkill.

After trying various explorations, I think @dlech's original suggestion of just enabling the keypad class might just be the cleanest (as opposed to adding something new for the one-button case).

We can add it in this way:

```diff
class CityHub:
    """LEGO® City Hub."""

    battery = _Battery()
    light = _ColorLight()
+   button = _Keypad((_Button.CENTER,))

```

I no longer think this is overkill: Because ``(Button.CENTER,)`` is true-ish, users can still simply do:

```python
if hub.button.pressed():
    print("yes")
```

This also lets us keep the `KeyPad.set_stop_button` method consistent on all hubs. It could look like this:

```python
class Keypad:
    """Get status of the buttons."""

    def __init__(self, available_buttons):
        pass

    def pressed(self):
        """Checks which buttons are currently pressed.

        :returns: Tuple of pressed buttons.
        :rtype: Tuple of :class:`Button`

        """
        pass

    def set_stop(self, buttons=(Button.CENTER,)):
        """Sets which button combination should cause a program to exit using
        a ``SystemExit`` exception.

        Arguments:
            buttons (tuple): The desired button combination.
        """
```

Defined in this way, the tuple format is identical to that returned by ``pressed()``. This can even be used to set button combinations ``(Button.CENTER, Button.BLUETOOTH)`` or no buttons at all `()`.

All this goes without additional implementation complexity since the `pbdrv` button flag is already a bitwise flag representing combinations.

dlech, 2021-03-17:
FYI, I don't think `hmi.c` is the right place to hook this in. I had a look at it and it just controls power off. 

Instead, I'm thinking it would be better to add a `pbsys_user_program_poll()` in `user_program.c` that is a copy of `pbsys_hmi_poll()` that gets modified to call `pbsys_user_program_stop()` based on the button state. Then the existing call to `pbsys_user_program_stop()` can be removed from the platform-specific `sys.c` files, and `pbsys_user_program_poll()` added to the existing poll functions in the same `sys.c` files.

laurensvalk, 2021-03-17:
That sounds good to me. I won’t be merging my commit any time soon, so feel free to make this change if you want to. I’ll just add the change at the Python module level later.

dlech, 2021-03-17:
Ok, will do

dlech, 2021-03-17:
One minor issue with adding this to the Keypad class is that it doesn't make sense for non-hub devices, like a remote control that might use the Keypad class as well.

laurensvalk, 2021-03-18:
> One minor issue with adding this to the Keypad class is that it doesn't make sense for non-hub devices, like a remote control that might use the Keypad class as well.

This should be okay since we don't really document the `KeyPad` class as a whole, just selected methods. We do the same for `ColorLight`, which doesn't have the `pattern` method exposed on all instances.

If we want to make intellisense aware of this we could internally just let `KeyPad(SimpleKeyPad)` to ensure only the right instances will have the `set_stop_button` method.

dlech, 2021-06-30:
Implemented in the commits above.

closed 2021-06-30 by dlech