pybricks/support #275: [Bug] Empty page on Pybricks Code on Android

laurensvalk, 2021-03-19:
**Describe the bug**
Pybricks Code gives an empty page on a new Android 10 Phone. This is the first time opening that page on this device.

It loads just fine on my previous Android phone and PC. It appears that the favicon does load, so something is getting through at least.

**To reproduce**
Steps to reproduce the behavior:
1. Buy and set up a new phone :smile: 
2. Go to code.pybricks.com
3. Get an empty page.

**Expected behavior**
Pybricks Code should load.

**Screenshots**
I'll add one in just a moment.

dlech, 2021-03-19:
Did you update all of the apps first? I got a chromebook and it did the same thing because it was on Chrome 84 or something like that. You can also enable ADB and connect it to Chrome on your computer to look at the console log to see what is going on.

laurensvalk, 2021-03-19:
Looks like that was it, thanks!

closed 2021-03-19 by laurensvalk