pybricks/support #283: [Bug] pybricksdev is very slow to run a script of 200 lines

Novakasa, 2021-03-27:
**Describe the bug**
Coming from https://github.com/pybricks/support/discussions/282, I measured the time for `BLEPUPConnection.run(script, wait=False)` to complete to be close to 10 seconds. This seems excessive for the task. Running the same script using Pybricks Code takes on average 2-3 seconds for me.

**To reproduce**
Run the following script for a PoweredUp City hub:
```python
import time
import asyncio
from pybricksdev.connections import BLEPUPConnection
from pybricksdev.ble import find_device

async def main_coro():
    device = await find_device("Pybricks Hub")
    hub = BLEPUPConnection()
    await hub.connect(device)
    t0 = time.time()
    await hub.run("autonomous_train.py", wait=False)
    t = time.time()-t0
    print(f"starting program took {t} seconds!")

asyncio.run(main_coro())
```
The output gives me:
```
Searching for Pybricks Hub
Connecting to 90:84:2B:BE:1E:40: Pybricks Hub
Connected successfully!
starting program took 9.359002113342285 seconds!
```
**Expected behavior**
Given that running the same script using Pybricks Code takes just about 2-3 seconds, I would expect the process to be as fast using `pybricksdev`

laurensvalk, 2021-03-28:
We are revising the program download process extensively at the moment. Once that is complete, `pybricksdev` can be updated to match, at which point we could look at the download speed as well.

laurensvalk, 2021-03-31:
It has also not been tested in notebooks and with multiple programs yet, so things might be a bit flaky for now. It probably also doesn't work with Move Hub just yet.

Feel free to share issues if you run into them.

Novakasa, 2021-04-01:
Great, thank you!

I checked out latest master commit, flashed to the most recent build using Pybricks Code, and adapted the above example to `PybricksHub`. This time, I measured a running time of just over 6 seconds. That's a great improvement!

I wanted to test running the same script using Pybricks Code now, but it doesn't seem to work with this firmware yet, I'm getting `Incompatible .mpy file' in the Pybricks Code console right after connecting, and the run button is greyed out even when connected to the hub and only using a hello world script. However, running on another hub with older firmware using Pybricks Code also takes about 6 second when manually timing it. So the run time from pybricks code seems to be matched now!

I will need to update my scripts for any changed pybricks API now, as it seems to quit immediately now. (I need to work on making any errors show up for me)

dlech, 2021-04-01:
Hopefully we will update `https://code.pybricks.com` within the next week, but if you are really anxious, you can build and run Pybricks Code locally (instructions are in the GitHub repo with the source code).

dlech, 2021-04-09:
> This time, I measured a running time of just over 6 seconds.

I assume this includes connecting, running and disconnecting the hub. If you just connect once and stay connected, running should be much faster.

Novakasa, 2021-04-10:
@dlech No, for testing the pybricksdev download speed I actually used the same method of timing it as outlined in the first post of this issue.

I just tested again using the updated pybricks code and measured 5.5 seconds for a program that I adpated to the new firmware. I timed it from clicking on the run button on pybricks code until the "Running downloaded program..." is printed.
The program I am downloading (though that should probably not matter at all): https://pastebin.com/snvpWVU5

I will probably test it on my laptop soon as well.

Novakasa, 2021-04-10:
I just tested with pybricks code on my laptop using it's built in bluetooth adapter, also came out as around 6 seconds.

I tested on both my Desktop PC and on my laptop on Chrome and on Microsoft Edge, in all cases the results were consistent.

Both my machines are Windows 10, but I have also Linux installed, I might try that soon.

Novakasa, 2021-04-10:
As a comparison, I also measured the time it takes to flash the firmware. I started the measurement when the progress bar in pybricks code shows up and stopped when it is complete. Came out as 73 seconds.

Novakasa, 2021-04-10:
I tested the download speed on my Linux Mint install now. I used pybricks code in Chrome.
Interestingly, I pretty consistently measure 4.5 seconds, which is on average 1 second faster than on windows on the same machine.

I also want to note that on all the configurations I tested today I got the "incompatibe .mpy file" error from time to time, which I could only fix by restarting the browser.

dlech, 2021-04-12:
I'm getting similar download times for the pasebin program, so I think we can close this unless pybricksdev is still way slower than Pybricks Code.

Novakasa, 2021-04-12:
No, the different download times were fixed with last week's changes.
Thanks for the confirmation of the download times!

closed 2021-04-12 by Novakasa