pybricks/support #284: [Feature] Remotely starting a stored program

Novakasa, 2021-03-27:
**Is your feature request related to a problem? Please describe.**
I am incorporating multiple PoweredUp Hubs in a larger project which is controlled centrally on a PC using `pybricksdev`. When running it, I have to run `BLEPUPConnection.run()` for each Hub, which for me takes about 10 seconds (see #283). Given that we have the possibility to store a program on the hub permanently, it would be a nice feature to be able to start that program remotely using `pybricksdev`, so the program will not have to be copied to the hub every time I am testing my larger project. Even when #283 is addressed, each download and run operation takes about 2-3 seconds for me even with Pybricks Code, so I still think that this would be a useful addition to `pybricksdev' especially if you are using scripts of large sizes, which might take a bit longer to copy still.

**Describe the solution you'd like**
I can imagine adjusting the API that you could seperate copying and starting a program like so:
```python
hub = BLEPUPConnection()
await hub.connect(my_address)
await hub.copy_script("my_script.py")
await hub.run_script("my_script.py")
```
And this should also work when ommitting `copy_script` when a program was already stored permanently when flashing the firmware.
If there might ever be the possiblity of storing multiple scripts on the hub, one could introduce a separate name parameter for `copy_script` and `run_script` so the scripts can be identified.

laurensvalk, 2021-07-30:
> It would be a nice feature to be able to start that program remotely using pybricksdev, so the program will not have to be copied to the hub every time I am testing my larger project.

Although seemingly unrelated, this would be solved by https://github.com/pybricks/support/issues/408, by running a script that just does `import main.py`.

And this is perhaps even more useful than a special start command, since you'd be completely flexible in passing additional commands. You could upload the same `main.py` to all trains, and then start each with its own parameters as needed:

```python
from main import SmartTrain

train = SmartTrain(param1, param2)
train.start()
```

laurensvalk, 2022-08-23:
This is now theoretically possible with https://github.com/pybricks/pybricks-micropython/pull/109 and subsequent PRs. Sending program size 0 is the same as pressing the button.

Although it might be nicer to make a command for it just like we have the stop command. Then we can get rid of the special size to trigger the REPL as well.

Novakasa, 2023-04-03:
> This is now theoretically possible with [pybricks/pybricks-micropython#109](https://github.com/pybricks/pybricks-micropython/pull/109) and subsequent PRs. Sending program size 0 is the same as pressing the button.
> 
> Although it might be nicer to make a command for it just like we have the stop command. Then we can get rid of the special size to trigger the REPL as well.

Could that currently work using pybricksdev?

closed 2023-04-28 by dlech

Novakasa, 2023-04-28:
If I understand correctly, the new function `start_user_program` indeed does what is requested here, however it seems that it doesn't really replace the old `run` function, since a lot of setup and state management for stdin, stdout, logfiles, etc still is only done in `run` (which always redownloads the program`. Is `start_user_program` supposed to be called in isolation?

For me, the ideal API would be to have `run` but just have the download be optional.