pybricks/support #287: Issue after update

thesynman, 2021-04-03:
Hi,
Pybricks has been working great for me - until today :(.

I've mostly been using pycharm and pybricksdev for development but today I wanted to try something quickly so I opened code.pybricks.com. It told me there was a new version, so I reloaded to update. When I connected to my Technic hub it told me it couldn't get device information and suggested checking that the firmware was up to date. So I reloaded the firmware. No error messages, but at the end the red/green/blue lights were still flashing, and I don't remember it ending like that before.

Anyway, I can connect to the hub but the repl and run buttons remain disabled. However, if I then press the hub button I see this in code.pybricks.com:

```
Running program from flash memory...
('technichub', '3.0.0b3', '6c3daec on 2021-04-02')
Hello, World!

...done.
```

and the repl and run buttons then become enabled.

If I try using pybricksdev I get this:

```
C:\pybricks\pybricksdev>poetry run pybricksdev run ble "Pybricks Hub" "print('hi there')"
Searching for Pybricks Hub
Connecting to 90:84:2B:5E:BE:7B: Pybricks Hub
Connected successfully!
2021-04-03 08:58:39,381:    INFO: Program size: 45 bytes
2021-04-03 08:58:39,489:    INFO: Sending: 100%

Running downloaded program...
hi there

...done.
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "C:\pybricks\pybricksdev\pybricksdev\cli\__init__.py", line 326, in main
    loop.run_until_complete(subparsers.choices[args.tool].tool.run(args))
  File "C:\Users\John Daniels\AppData\Local\Programs\Python\Python38\lib\asyncio\base_events.py", line 616, in run_until_complete
    return future.result()
  File "C:\pybricks\pybricksdev\pybricksdev\cli\__init__.py", line 154, in run
    await hub.disconnect()
  File "C:\pybricks\pybricksdev\pybricksdev\ble.py", line 149, in disconnect
    await self.client.disconnect()
  File "C:\pybricks\pybricksdev\.venv\lib\site-packages\bleak\backends\dotnet\client.py", line 281, in disconnect
    await asyncio.wait_for(event.wait(), timeout=10)
  File "C:\Users\John Daniels\AppData\Local\Programs\Python\Python38\lib\asyncio\tasks.py", line 501, in wait_for
    raise exceptions.TimeoutError()
asyncio.exceptions.TimeoutError

C:\pybricks\pybricksdev>

```

Any suggestions?

thesynman, 2021-04-03:
In case it helps diagnose this problem, I've discovered that the behaviour using pybricksdev isn't consistent. Sometimes it more-or-less works:
```
C:\pybricks\pybricksdev>poetry run pybricksdev run ble "Pybricks Hub" "print('hi there')"
Searching for Pybricks Hub
Connecting to 90:84:2B:5E:BE:7B: Pybricks Hub
Connected successfully!
2021-04-03 09:27:29,836:    INFO: Program size: 45 bytes
2021-04-03 09:27:29,904:    INFO: Sending: 100%

Running downloaded program...
hi there

...done.

C:\pybricks\pybricksdev>
```
Sometimes I get this:
```
C:\pybricks\pybricksdev>poetry run pybricksdev run ble "Pybricks Hub" "print('hi there')"
Searching for Pybricks Hub
Connecting to 90:84:2B:5E:BE:7B: Pybricks Hub
Connected successfully!
2021-04-03 09:27:02,073:    INFO: Program size: 45 bytes
2021-04-03 09:27:02,184:    INFO: Sending: 100%
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "C:\pybricks\pybricksdev\pybricksdev\cli\__init__.py", line 326, in main
    loop.run_until_complete(subparsers.choices[args.tool].tool.run(args))
  File "C:\Users\John Daniels\AppData\Local\Programs\Python\Python38\lib\asyncio\base_events.py", line 616, in run_until_complete
    return future.result()
  File "C:\pybricks\pybricksdev\pybricksdev\cli\__init__.py", line 153, in run
    await hub.run(script_path, args.wait == 'True')
  File "C:\pybricks\pybricksdev\pybricksdev\connections.py", line 307, in run
    await self.send_message(chunk)
  File "C:\pybricks\pybricksdev\pybricksdev\connections.py", line 268, in send_message
    raise ValueError(
ValueError: Expected checksum 96 but received 114.

C:\pybricks\pybricksdev>
```

laurensvalk, 2021-04-03:
Pybricksdev has been updated also, to match the changes in the firmware. Are you already using the latest version?

laurensvalk, 2021-04-03:
Regarding the run and repl buttons not working in the IDE, is this issue persistent? Could you try CTRL+F5 to perform a hard refresh of the page?

thesynman, 2021-04-03:
I hadn't updated pybricksdev - now done. The behaviour is different but still inconsistent. Once it worked OK, but I mostly get something like this:
```
C:\pybricks\pybricksdev>poetry run pybricksdev run ble "Pybricks Hub" "print('hi there')"
Searching for Pybricks Hub
Downloading 45 bytes in 1 steps.
Progress: 100%
2021-04-03 10:23:15,249: WARNING: Expected checksum 96 but got 13
2021-04-03 10:23:15,249: WARNING: Expected checksum 96 but got 32
e

...done.
```
and it doesn't exit.

Or it will run ok but not exit. It was rock-solid before the update. There's definitely something wrong.

As for pybricks-code, the first time I got this:
[image]
The second time I connected it worked. But after that it's the same behaviour I reported before: the hub connects but the run and repl buttons aren't enabled until I press the button on the hub.

BertLindeman, 2021-04-03:
Using win10 Microsoft Edge Versie 89.0.774.68 (Officiële build) (64-bits)

No working run and repl buttons. 
The connect button is also very slow (3 a 4 seconds).
Waited a few minutes after successful connect but no run/repl.

```Pybricks Code v1.0.0-beta.8``` and CTRL-F5  no success.

Last night I sometimes had a working connection and now not.

Using ```Google Chrome Version 89.0.4389.114 (Official Build) (64-bit)``` 
did only succeed the second CTRL-F5 (while connected)

Must be a timeout somewhere?

BertLindeman, 2021-04-03:
Forgot to tell: running on a technic hub, but the movehub feels similar.

Can confirm that a short press on the hub button starts to run the program on my code.pybricks.com screen.

Some time after "connect":
[image]

After hub-button press:
[image]

laurensvalk, 2021-04-03:
Thank you both for testing so thoroughly! There have been some big revisions on the Bluetooth side and it looks like we've still got some work to do.

Meanwhile, @thesynman, in case you want to use an older version of `pybricksdev`, you can, by reverting to a [previous firmware version](https://docs.pybricks.com/en/latest/start_pup.html#installing-the-latest-build).

thesynman, 2021-04-03:
@laurensvalk I've already reverted to version `v4.6.0+v3.0.0b2-1-g0072f4ec` of the firmware and to commit `7a6d2f4` of pybricksdev (it might be a good idea to put a tag on that commit so that others can more easily find a working combination).

Thanks for your quick responses, and I hope you find the problem.

dlech, 2021-04-03:
> I can connect to the hub but the repl and run buttons remain disabled.

Hmm... I fixed this already. The problem is that Pybicks Code is not getting the initial status update from the hub. Looks like the common thread here is running on Windows, so I will try to reproduce there.

As you have found, the workaround is to press the hub button a couple of times.

dlech, 2021-04-03:
> v4.6.0+v3.0.0b2-1-g0072f4ec

Where did you get this version from?

thesynman, 2021-04-03:
@dlech Yes, sorry, that is the version number - I added the v4.6.0 myself because that was mentioned on the build. I'm unclear about how your version numbering works.

dlech, 2021-04-03:
> Hmm... I fixed this already. The problem is that Pybicks Code is not getting the initial status update from the hub. Looks like the common thread here is running on Windows, so I will try to reproduce there.

I have confirmed this seems to be a Chromium Windows bug and filed an issue. https://bugs.chromium.org/p/chromium/issues/detail?id=1195592

So we need to change the firmware to work around this. (We need to do this anyway, since the notification could be missed due to, e.g. a noisy RF environment, not just this specific Chromium bug.)

I also found additional Bluetooth issues on Windows, so we have reverted Pybricks Code to the previous version for now. (In addition to the GATT error message posted above, I saw one other error message and the Technic hub itself doesn't seem to disconnect when clicking the Bluetooth button in Pybricks code to disconnect)

dlech, 2021-04-05:
I just pushed v1.0.0-beta.9 with a fix. Can you please update and test again?

BertLindeman, 2021-04-05:
@dlech just updated code.pybricks.com oke:
```
Pybricks Code v1.0.0-beta.9
```

Switched OFF ```include current program ``` and
downloaded the corresponding firmware using the download firmware button in 
```Microsoft Edge Versie 89.0.774.68 (Officiële build) (64-bits)```

After connecting the buttons "Download" and "REPL" are correctly activated.
After pressing the run button I get:
```
GValueError: incompatible .mpy file
``` The character before "Value" changes some of the times.

If I replace the program in the editor with a simple program like this:
```
from pybricks import version
print(version)
```
Then the run succeeds:
```
Running downloaded program...
('technichub', '3.0.0b4', '1a02dfb on 2021-04-05')

...done.
```

But if I add ```from pybricks.tools import StopWatch``` to the program . . . .
The incompatible mpy error returns.

So some success, but not completely yet?

dlech, 2021-04-05:
I started #290 already for incompatible .mpy issue already. Seems to just be a problem with Edge for some reason.

BertLindeman, 2021-04-06:
Also tried chrome:
```
Google Chrome is up to date
Version 89.0.4389.114 (Official Build) (64-bit)
```

So THIS item can be closed?

laurensvalk, 2021-04-06:
Before we close this, I'd be curious if @thesynman's issue:

> If I try using pybricksdev I get this [...]:

has gotten any better with the latest update.

thesynman, 2021-04-06:
I can confirm that pybricks-code now seems to work ok, but it's hard to be sure because of #290.

Win 10 and Chrome Version 89.0.4389.114 (Official Build) (64-bit)

@laurensvalk I've pulled the latest from pybricksdev and still seeing the same sort of errors (with the latest firmware). I'm doing:
`poetry run pybricksdev run ble "Pybricks Hub" "print('hi there')"`
and the outcome is inconsistent. Either a) (rarely) seems to work but the program never exits, b) I see checksum errors but the program runs but doesn't exit, c) I see checksum errors and then `Error during program download`.

In short, pybricksdev is still not working for me.

laurensvalk, 2021-04-06:
Thanks for confirming! I'll move it to a dedicated issue so we can look at it later.

closed 2021-04-07 by dlech