pybricks/support #290: [Bug] ValueError: incompatible .mpy file on Microsoft Edge Browser Pybricks Code v1.0.0-beta.9

dlech, 2021-04-05:
**Describe the bug**
What is the problem?

Clicking the download and run button fails with: ValueError: incompatible .mpy file

**To reproduce**
Steps to reproduce the behavior:
1. Open Pybircks Code in Microsoft Edge on Windows 10
2. Connect to a Technic hub (update firmware if needed)
3. Create a program
4. click download and run

**Expected behavior**
What did you expect to happen instead?

It should work the same as Chrome browser.

**Screenshots**
There is a saying that a picture is worth a 1000 words. Screenshots really help to identify and solve problems.

BertLindeman, 2021-04-06:
Tried ```Google Chrome is up to date
Version 89.0.4389.114 (Official Build) (64-bit)```
with this program:
```
print("Just print version; program added to firmware")
from pybricks import version
print(version)
```
And got:
```
 ValueError: incompatible .mpy file
```

Reduced the program by commenting out the first print line:
```
# print("Just print version; program added to firmware")
from pybricks import version
print(version)
```
 and got:
```
o
Running downloaded program...
('technichub', '3.0.0b4', '1a02dfb on 2021-04-05')

...done.
```

No idea where the leading "o" comes from . . . .

If I make the first print line shorter it **does** work.
E.G. ```print("Just print version")```

No idea if it **is** related, messages from developer tools:
[image]

thesynman, 2021-04-06:
Same problems here using Win 10 and Chrome Version 89.0.4389.114 (Official Build) (64-bit)

dlech, 2021-04-06:
> No idea where the leading "o" comes from . . . .

It is a checksum value that isn't supposed to be sent to the terminal.

dlech, 2021-04-06:
The fix for #287 caused this issue. v1.0.0-beta.10 is published now which should fix this issue.

BertLindeman, 2021-04-06:
using updated beta-10 running repeated this program (kept pressing "run"):
```
```
got this result (trimmed a few identical entries from the top):

dlech, 2021-04-06:
So we can call it significantly improved, but not entirely fixed? :smile:

BertLindeman, 2021-04-06:
Sure: no longer stable error. Another error?

Also occurs on Win10  Google Chrome.
disconnect / connect does not resolve this situation.

CTRL-F5 does resolve the problem.
Both browsers.
But problem can be recreated by running the program a few times (twice was enough now)

dlech, 2021-04-07:
An additional fix is made in v1.0.0-beta.11.

dlech, 2021-04-07:
I think I found yet another race condition. If the program ends too quickly, we miss the user program running status flag, and Pybricks Code gets stuck with the run button disabled.

dlech, 2021-04-07:
> I think I found yet another race condition caused by the fix I just made. If the program ends too quickly, we miss the user program running status flag, and Pybricks Code gets stuck with the run button disabled.

Should have opened a new issue for this, but it is fixed now in v1.0.0-beta.12

Novakasa, 2021-04-11:
As mentioned in https://github.com/pybricks/support/issues/283#issuecomment-817160202, I still had `ValueError: incompatible .mpy file` show up for me sometimes on both Windows and Linux, Chrome and Edge. Once it showed up, I could only fix it by restarting the browser. I used Pybricks code beta.12 with the firmware that it installs by default.

dlech, 2021-04-11:
If you can figure out exact steps to reproduce the problem in beta 12, that would be helpful.

dlech, 2021-04-12:
> Once it showed up, I could only fix it by restarting the browser

Additional fixes have been made in beta 13, so hopeful that makes it so we can't get in a bad state that requires a browser restart.

Novakasa, 2021-04-12:
I did some superficial testing just now, I didn't update pybricks code just yet (had the "restart to update" popup showing).
On chrome I started and stopped my program a few times. At some point in this process, the hub seemingly crashes (continuous LED, not responsive to button presses). When that happens I am always forced to shut down the hub via long pressing the button, which still works. Upon turning it on again, I could reconnect and start the program without problem.
Upon switching to Edge, I did the same, starting and stopping my program a few times. At some point the hub similarly freezed up again. Upon restarting the hub there, I had the `incompatible .mpy file` error show up. Like previously, this persists even when reconnecting the hub.

dlech, 2021-04-12:
> I didn't update pybricks code just yet (had the "restart to update" popup showing).

What was the exact version used for this test?

Novakasa, 2021-04-12:
> What was the exact version used for this test?

That was beta.12.

Now I just tested it on Edge using beta.13. After connecting, I started the program (the same as in the pastebin from the other issue), which runs without issue. After pressing the stop button on pybricks code, the hub became unresponsive. When pressing run after that, it is greyed out for 1-2 seconds and then is enabled again, but no message shows up in the terminal. I could reproduce it one more time, but now I tried a few more times and starting and stopping seems to work without problem now. I will have more time for testing tomorrow.

dlech, 2021-04-12:
Thanks. Sounds like we finally (:crossed_fingers:) got the Pybricks Code issues sorted out. The hub freezing sounds like it could be a memory corruption issue in the firmware.

dlech, 2021-04-13:
It is still possible to get `ValueError: incompatible .mpy file` if you press a key when the terminal is in focus and a program is not running(#303) and if that same error message is caused by something else, it is probably a new bug, so closing this one.

closed 2021-04-13 by dlech