pybricks/support #291: [Bug] Technic hub stays "connected" after "Disconnect Bluetooth" in win10 Chrome

BertLindeman, 2021-04-06:
**Describe the bug**
In code.pybricks.com on win10 in Google Chrome 
After press on "disconnect Bluetooth" the hub does not disconnect
The Bluetooth button **does** show it is disconnected

**To reproduce**
Steps to reproduce the behavior:
1. Open Pybricks Code in Google Chrome on Windows 10
2. Connect to a Technic hub
3. Click "Disconnect Bluetooth"
4. Observe the blue light on  the hub: it keeps steady blue
5. have to power off / on the hub to reconnect

**Expected behavior**
The light on the hub switch to blinking blue to show it wants to connect / pair 

**Screenshots**
None now

**versions**
- Windows 10
- Google Chrome Version 89.0.4389.114 (Official Build) (64-bit)
- Pybricks Code v1.0.0-beta.9
- technichub
- Firmware:  '3.0.0b4', '1a02dfb on 2021-04-05'

BertLindeman, 2021-04-06:
Problem **not** seen on Microsoft Edge

dlech, 2021-04-06:
I was seeing this too last week but could not reproduce it yesterday using v1.0.0-beta.9/v3.0.0b4.

Could you do a Bluetooth capture when this happens? https://bleak.readthedocs.io/en/latest/troubleshooting.html#capture

Does rebooting the computer make the problem go away?

BertLindeman, 2021-04-06:
> 
> 
> I was seeing this too last week but could not reproduce it yesterday using v1.0.0-beta.9/v3.0.0b4. (Pretty sure I saw it in Edge though.)
> 
> Could you do a Bluetooth capture when this happens? https://bleak.readthedocs.io/en/latest/troubleshooting.html#capture
> 
> Does rebooting the computer make the problem go away?

Will try, may take some though. But not weeks ;-)

BertLindeman, 2021-04-06:
Without booting win10 tried a few times both on Google Chrome and Edge.
and ```('technichub', '3.0.0b4', '1a02dfb on 2021-04-05')```

Problem did not re-occur. 

"did" some tests resulting in 
```ValueError: incompatible .mpy file```

Tested just in time with Pybricks Code v1.0.0-beta.9 before it would update to 10.

As beta-10 is on its way, closing this issue.
If needed we can re-open

closed 2021-04-06 by BertLindeman

dlech, 2021-04-08:
I just had a similar issue with `pybricksdev` not disconnecting and rebooting fixed it. So I think the Windows Bluetooth stack just gets in a bad state somehow.