pybricks/support #293: Robot-Inventor: Activate recovery mode Linux - SecurityError: Access denied.

scotus-49, 2021-04-06:
I am trying to use the recovery mode to rescue my Robot Inventor Hub. I have followed the extra instructions for Linux, and downloaded and copied the udev rules files to /etc/udev/rules.d/99-pybricksdev.rules. 

I am still getting the SecurityError: Access denied message. Can you help?

laurensvalk, 2021-04-06:
Which browser do you use? I had the same with Chromium on a more recent Linux build.

It did work in Google Chrome, I think.

But since you're on Linux, you can also do:

```
sudo apt install dfu-util
dfu-util -a 0 -s 0x08008000 -D /path/to/backup.bin
```

scotus-49, 2021-04-06:
Thanks for the help. I was using Chromium, and after installing Google Chrome it seemed to work. I got the following messages:

Preparing installation.
Installing firmware.
Starting hub.
Done!

Then the Bluetooth button went dark, and the Hub is completely unresponsive - playing dead. I used the file 96112deb24a934bfc19a13c7ea620f54.bin {storm - v1.0.06.0034-b0c335b - 2.1.4.13} downloaded from the linked site. What should I do now?

laurensvalk, 2021-04-06:
What is the size of file you downloaded? ~100Kb by any chance? I think I might have just done the same as you, right-clicking and save-as. But that gives you an HTML file, not the .bin file.

So be sure to click on the file and then click download, which gives you [this link](https://github.com/gpdaniels/spike-prime/raw/master/firmware/storm%20-%20v1.0.06.0034-b0c335b%20-%202.1.4.13/96112deb24a934bfc19a13c7ea620f54.bin).

That file is about 500kb. Just run the procedure again with the new file.

Let me know how it goes.

scotus-49, 2021-04-06:
Thanks again Laurens - it worked this time! I should have seen that the icon of my downloaded file was showing that it was a text file - about 115K.

Now to my windows machine to reload the Pybricks using the Lego app...

laurensvalk, 2021-04-06:
Or, try this page so you don't have to use Windows :wink: 

https://beta.pybricks.com/

scotus-49, 2021-04-06:
Thanks for the link - I've used it to install the firmware, and it all went fine.  

However, with my newly installed firmware, I'm having problems connecting to Pybricks Code. Whether I use Windows or Linux - I'm getting:
"Connected to hub but failed to get Device Information dervice.
Ensure that you are using the most recent firmware..."

I am assuming that in Pybricks mode the InventorHub bluetooth button is inactive.

laurensvalk, 2021-04-06:
Yeah, we're doing quite extensive Bluetooth revisions at the moment. Hopefully most of those will be fixed this week.

I'll keep this issue open as a reminder that I upload the latest firmware to this installation page.

scotus-49, 2021-04-06:
Thanks for the update. I thought I'd done something wrong again...

Should I again use your page https://beta.pybricks.com/ to update the firmware, or should I wait a bit?

closed 2021-04-06 by dlech