pybricks/support #295: [Feature] Show the version of the firmware in the popup of the download button

BertLindeman, 2021-04-07:
**Is your feature request related to a problem? Please describe.**
Would be nice to see the version of the firmware before the actual download

**Describe the solution you'd like**
Show the version of the firmware in the popup of the download button

**Describe alternatives you've considered**
Download the firmware and run to see what version it has :smile:
Takes some time we want to avoid spending...

**Additional context**
[image]

**Additional thought**
Only if very easy to implement and automatically filled in after that.

dlech, 2021-04-07:
Also see https://github.com/pybricks/support/issues/173#issuecomment-812752509

BertLindeman, 2021-04-07:
Ah thanks David, my memory is not as it was.
I even commented on that 😞 

In that case ( #173 ) this feature is not needed.

Closing

closed 2021-04-07 by BertLindeman