pybricks/support #298: [Feature] Improve

laurensvalk, 2021-04-12:
**Is your feature request related to a problem? Please describe.**

This is overly long and pollutes the limited space in the output window.

```

Running downloaded program...

...done.
```

I believe it was intended to clarify when programs start and end, but I don't see this being any more helpful than a simple separator:

[image]

**Describe the solution you'd like**

A simple separator should suffice `--------------------------------------`

**Fancier solution**

Instead of letting the firmware something, we could have Pybricks Code insert something like

`Program started  at 04-12-2020 12:08:144`
`Program stopped at 04-12-2020 12:13:134`

dlech, 2021-04-12:
> Instead of letting the firmware something, we could have Pybricks Code insert something

I agree this would be nice, but we currently have enough race condition problems with starting and stopping the program.

> A simple separator should suffice `--------------------------------------`

I think 3 dashes would be enough.

laurensvalk, 2021-04-12:
Or none at all, assuming we can get the download and run buttons to work well enough.

dlech, 2021-04-12:
> Or none at all,

doing this for now

laurensvalk, 2021-04-13:
This looks good, so closing.

closed 2021-04-13 by laurensvalk