pybricks/support #300: [Bug] Pybricks Code only works in landscape mode on Android

laurensvalk, 2021-04-12:
**Describe the bug**

I can reproduce some of the download and run issues on Android, but my keyboard is getting in the way :smile: 

Forced landscape mode makes it very hard to use. See screenshot below. I have Android 11.

**To reproduce**

Save Pybricks Code as an app on a modern Android device.

**Expected behavior**
Rotate the screen when rotating the phone

**Screenshots**

laurensvalk, 2021-04-12:
Having the keyboard open at all times might also be useful, to avoid jumping around much.

dlech, 2021-04-12:
I did that intentionally because there isn't room for the buttons the other way and they cover the code. I think it would be better to just set a minimum screen size of 10 or 11 inches and not try to support smaller devices.

laurensvalk, 2021-04-12:
Having the buttons max-button size scaled to fit could work. They're still big enough that way.

It's definitely not critical, but because everything else is working so well on Android, it almost seems worth it.

closed 2021-12-27 by dlech