pybricks/support #301: [Bug] Deadlock when pressing start and pressing button simultaneously

laurensvalk, 2021-04-12:
**Describe the bug**

Buttons remain grayed out after clicking the Technic Hub button and the Pybricks Code play button at the same time.

This happens 100% of times for me on Ubuntu 18.04 and on Android 11.

**To reproduce**
See program in screenshot below. Then press  Technic Hub button and the Pybricks Code play button at the same time.

**Expected behavior**
Either one of the programs to run successfully, or nothing at all; at least no deadlock.

It's a bit of an edge case, but it may help us uncover some of the other issues we are seeing.

**Screenshots**

It happens whichever you press first.

This happens usually when physical button is pressed first:

[image]

The other way around, you can usually get to this state:
[image]

dlech, 2021-04-12:
Should be fixed in beta 13

dlech, 2021-04-12:
There is still the edge case where pressing the hub button during download "cancels" the download and prints `ValueError: incompatible .mpy file` but the buttons don't get stuck anymore.

laurensvalk, 2021-04-13:
> Should be fixed in beta 13

Confirmed.

> There is still the edge case where pressing the hub button during download "cancels" the download and prints ValueError: incompatible .mpy file but the buttons don't get stuck anymore.

Confirmed, and this sounds fine to me.

closed 2021-04-13 by laurensvalk