pybricks/support #302: [Bug] Deadlock when connecting to bootloader just too late

laurensvalk, 2021-04-13:
**Describe the bug**

If you don't keep pressing the Technic Hub button long enough, connecting fails (as expected), but the button state remains locked.

Then after a while (30 sec?), you get an unexpected error:

**To reproduce**

- For convenience, erase the firmware first (e.g. make a failed download).
- This lets you activate the bootloader with a single click.
- Click flash button, select LEGO Bootloader
- Be just too late to press connect. The browser still lets you, despite the hub already being off.

**Expected behavior**
Failed install, but ability to try again without hard refresh

**Screenshots**
[image]

dlech, 2021-04-13:
Operating system?

laurensvalk, 2021-04-13:
Verified on Ubuntu 18.04 and android, using technic hub.

dlech, 2021-04-13:
In the screen shot, it looks like the button is enabled again after the error. So I think this is working as well as can be expected.

laurensvalk, 2021-04-13:
Sounds good to me.

closed 2021-04-13 by laurensvalk