pybricks/support #307: [Bug] Unresponsive stop button after idle for a few minutes

laurensvalk, 2021-04-14:
**Describe the bug**

I was working on [this demo](https://pybricks.com/projects/sets/technic/42099-off-roader/keyboard-remote/#the-remote-control-program). Usually, the hub and/or Pybricks Code got stuck, unable to stop, if I left it on for a few minutes while working on the article.

I think Pybricks Code recovered after manually shutting down the hub with a long button press, so no hard refresh was required.

**To reproduce**
Run [this demo](https://pybricks.com/projects/sets/technic/42099-off-roader/keyboard-remote/#the-remote-control-program). Press a few keys drive around. Then forget about the robot while you leave it running.

Drink a coffee and come back. Find Pybricks Code and the hub unresponsive.

``Pybricks Code v1.0.0-beta.14``

``Version 89.0.4389.90 (Official Build) Built on Ubuntu , running on Ubuntu 18.04 (64-bit)``

**Expected behavior**
Keep working after :coffee: 

**Screenshots**
[image]

dlech, 2021-04-14:
I have a feeling that this is the same sort of firmware problem as #304 and/or #232.

From the Pybricks Code side, there is currently nothing in the Pybricks protocol that acknowledges the receipt of a command. And it is possible for user programs to catch a `SystemExit` exception, so Pybricks Code shouldn't necessarily expect for a program to actually stop after the stop button is pressed.

laurensvalk, 2021-04-14:
> I have a feeling that this is the same sort of firmware problem

Same here, but I thought I'd keep reporting in order to get a minimal / reproducable case.

laurensvalk, 2021-04-26:
This should be fixed by #306.

closed 2021-04-26 by laurensvalk