pybricks/support #308: [Question] How to get the SPIKE prime hub blink like it should

BertLindeman, 2021-04-15:
**Question**
After testing for #304 I notice that the color of the hub button blinks has changed.

**Context**
Possibly multiple runs to create a test for #304
So far used beta.pybricks.com to reinstall the firmware. But no change there.
I would expect the pybricks firmware to start blinking blue until connected. But it is switching
"some other color" and blue.
Even more strange to me. . . . The dual boot into LEGO firmware does not blink white, but it is switching
"some other color" and white.
A thought I had was this related to "Include current program" but that would not be what I would expect of reinstalling the firmware.
I did no further actions to be able to debug this situation.
Anything I can do to reset the colors?

**Screenshots**
LEGO firmware awaiting Bluetooth connect:
https://user-images.githubusercontent.com/8142081/114867529-3a900f00-9df5-11eb-827c-54c594d74382.mp4

pybricks firmware awaiting Bluetooth connect:
https://user-images.githubusercontent.com/8142081/114867669-601d1880-9df5-11eb-8b21-a8b9ed8b6635.mp4

laurensvalk, 2021-04-15:
An orange blink means low battery.

This is a good one to add to the docs. Thanks for the reminder!

BertLindeman, 2021-04-15:
Rats, I should have thought of that.
THANKS Laurens!

Yesterday I had it at 100%, but those light tests use quite some juice.

Was a good call to make this a question issue.

So the answer is: "load the battery"

Closing......

BertLindeman, 2021-04-15:
Trailing thought:
The orange could be a bit more prominent.

laurensvalk, 2021-04-15:
I'll keep the issue open until I've added a page to the docs to explain all the light statuses, like low battery, but also Bluetooth, blinking etc.

dlech, 2021-05-12:
Lights are documented at https://pybricks.com/install/using-powered-up-hubs/ now (the URL will probably change as we organize the site).

closed 2021-05-12 by dlech