pybricks/support #312: [Bug] GATT Server is disconnected. Cannot retrieve services. (Re)connect first with `device.gatt.connect`.

Stari89, 2021-04-18:
**Describe the bug**
When installing the pybricks firmware to the TechnicHub, I get the "GATT Server is disconnected. Cannot retrieve services. (Re)connect first with `device.gatt.connect`." error message.
OS: Windows v 20H2 (OS Build 19042.928)
Browser: Chrome 89.0.4389.128 (Official Build) (64-bit)
BT: Qualcomm Atheros QCA61x4A Bluetooth 4.1
Log: [bth_hci.zip](https://github.com/pybricks/support/files/6331085/bth_hci.zip)
Same bug as #56 which is now closed.

**To reproduce**
Steps to reproduce the behavior:

1. Go to Pybricks Code.
2. Press and hold the hub button. Wait for a blinking pink light.
3. While you hold it, click the firmware update button.
4. Select the LEGO Bootloader and click Pair.
5. Wait until the light turns off, and then blinks red/green/blue.
6. Release the button and wait for the installation to finish.
7. Error message appears.

**Expected behavior**
The error message should not appear and I should have the pybricks firmware installed.

**Screenshots**
[image]

laurensvalk, 2021-04-18:
Thanks for reporting!

As a workaround for now, perhaps you can try the installation with an Adroid device. After that, you may be able to use your PC again to write scripts.

dlech, 2021-04-18:
Thanks for taking the time to log the Bluetooth. Here is what it looks like decoded:

```
1	0.000000	host	controller	HCI_CMD	10	Sent LE Set Scan Parameters
2	0.002231	controller	host	HCI_EVT	6	Rcvd Command Complete (LE Set Scan Parameters)
3	0.002247	host	controller	HCI_CMD	5	Sent LE Set Scan Enable
4	0.004149	controller	host	HCI_EVT	6	Rcvd Command Complete (LE Set Scan Enable)
5	0.248167	controller	host	HCI_EVT	35	Rcvd LE Meta (LE Advertising Report)
6	0.249198	controller	host	HCI_EVT	14	Rcvd LE Meta (LE Advertising Report)
7	0.365161	controller	host	HCI_EVT	31	Rcvd LE Meta (LE Advertising Report)
8	0.366164	controller	host	HCI_EVT	14	Rcvd LE Meta (LE Advertising Report)
9	0.613166	controller	host	HCI_EVT	45	Rcvd LE Meta (LE Advertising Report)
10	0.616163	controller	host	HCI_EVT	40	Rcvd LE Meta (LE Advertising Report)
11	0.719178	controller	host	HCI_EVT	45	Rcvd LE Meta (LE Advertising Report)
12	0.722167	controller	host	HCI_EVT	40	Rcvd LE Meta (LE Advertising Report)
13	1.668169	controller	host	HCI_EVT	45	Rcvd LE Meta (LE Advertising Report)
14	1.671166	controller	host	HCI_EVT	40	Rcvd LE Meta (LE Advertising Report)
15	2.133595	host	controller	HCI_CMD	5	Sent LE Set Scan Enable
16	2.135157	controller	host	HCI_EVT	6	Rcvd Command Complete (LE Set Scan Enable)
17	2.135174	host	controller	HCI_CMD	10	Sent LE Set Scan Parameters
18	2.137161	controller	host	HCI_EVT	6	Rcvd Command Complete (LE Set Scan Parameters)
19	2.155515	host	controller	HCI_CMD	28	Sent LE Create Connection
20	2.157158	controller	host	HCI_EVT	6	Rcvd Command Status (LE Create Connection)
21	2.397158	controller	host	HCI_EVT	21	Rcvd LE Meta (LE Connection Complete)
22	2.397175	host	controller	HCI_CMD	5	Sent Read Remote Version Information
23	2.408160	controller	host	HCI_EVT	6	Rcvd Command Status (Read Remote Version Information)
24	42.404219	controller	host	HCI_EVT	10	Rcvd Read Remote Version Information Complete
25	42.405688	localhost ()	LEGOSyst_86:68:ff (LEGO Bootloader)	ATT	11	Sent Exchange MTU Request, Client Rx MTU: 527
26	42.410214	controller	host	HCI_EVT	6	Rcvd Disconnect Complete
27	42.507490	host	controller	HCI_CMD	28	Sent LE Create Connection
28	42.509187	controller	host	HCI_EVT	6	Rcvd Command Status (LE Create Connection)
```

For some reason, it is disconnecting instead of finishing the MTU exchange.

When I have seen this problem in Windows, usually retrying again works. If it doesn't, rebooting Windows will clear up the problem.

Stari89, 2021-04-19:
Update: I tried in my company's laptop and it worked just fine. I could install the firmware and execute programs. Then I went back to my own PC and tried executing the same program from it. Same error message appeared.
Cheers

dlech, 2021-04-22:
What is different about the work computer? Windows version? Bluetooth chip?

bdhughes0047, 2021-05-12:
Ha!  Totally my bad, I was laser focused on the exact same subject error that I didn't notice the specific support site!  I've never even heard of Pybricks until now, but it looks fun!  Cheers!  B.

dlech, 2023-03-23:
I'm going to go ahead and close this one since it is getting a bit old and hasn't been updated in a while and we have other open issues with the same error message.

For any future visitors here with the same error message on Windows, it would be helpful to know which Bluetooth adapter you are using so we can try to figure out which ones are compatible and which ones have issues.

Please see https://github.com/pybricks/support/issues/921 for a PowerShell command line to get this info and post your results in that issue.

closed 2023-03-23 by dlech