pybricks/support #313: [Question] Loosing contact with hub running remote control on set 42099

BertLindeman, 2021-04-21:
**Question**
Running the code for keyboard remote for the 42099-off-roader.
Soon (5 to 15 keypresses) loses contact with the hub. 
- Hub keeps running 
- sometimes code.pybricks thinks to have lost Bluetooth contact, sometimes NOT.
- recovery via long-press on the hub is OKE.

Might this symptom be related to #304 ?

**Context**
Before I try more diagnosis, might be good to know if #304 indeed is the cause.
Also tested with the more limitted ```remote.py``` and also had similar problems.
But it took more time, so more keypresses to get stuck.

This occurred both on win10 and also on Linux mint.
Freshly loaded batteries
Hub:  ```MicroPython bf4bf0e on 2021-04-12; LEGO Technic Hub with STM32L431RC```

**Screenshots**
None so far.

dlech, 2021-04-21:
We've been tracking this in #306

BertLindeman, 2021-04-21:
Ah, thanks @dlech 
I did not make the connection :wink:

Indeed found in the developer info:
[image]

You already gave the dup label, so closing.
Sorry for the extra work.

dlech, 2021-04-21:
No worries. :smile: