pybricks/support #315: [Feature] Add busy state to bluetooth button while connecting

laurensvalk, 2021-04-26:
**Is your feature request related to a problem? Please describe.**
After clicking connect in the connection dialog, it can take several seconds on some PC's before the start button becomes enabled.

**Describe the solution you'd like**
It would be nice to indicate the busy state, just as during firmware install initialization.

**Describe alternatives you've considered**
Buy @martynboogaarts a new laptop :smile:

dlech, 2021-05-14:
The firmware update button also needs to be disabled during this time.

martynboogaarts, 2021-05-15:
understand

PS new DELL on its way ...

laurensvalk, 2021-05-15:
Don't throw the old one out, it seems especially good at finding issues in our app 😄

dlech, 2021-05-17:
This is done in RC2.

closed 2021-05-17 by dlech