pybricks/support #327: [Question] Hub light status and button grey out

vanessahsw, 2021-05-05:
I am trying to connect city hub and the firmware seems to be installed properly. Afterwards, I connect the hub via bluebooth and got a consistent blue solid light. 

However, the run and stop program buttons were greyed out so I was unable to execute anything. Just hoping to get some guidance on how to proceed forward. 

Thanks.

laurensvalk, 2021-05-05:
Thanks for reporting this. Which operating system and browser are you using?

dlech, 2021-05-05:
We also have a [Bluetooth troubleshooting guide](https://github.com/pybricks/support/discussions/270).

vanessahsw, 2021-05-07:
I am using Chrome - Version 87.0.4280.88 (Official Build) (x86_64) and Mac OS 10.10.5.

This is what I saw from the device log:
```
BluetoothEvent[22:32:29] BluetoothAPI: 0x7fe013d695b0
```

No error message was seen in the developer console.

dlech, 2021-05-07:
I have a hunch that the problem could be because of the older macOS version. Chrome system requirements say 10.11 minimum.

I am able to connect and use a city hub using Chrome 90 and macOS 11.

dlech, 2021-05-11:
> This is what I saw from the device log:
> 
> ```
> BluetoothEvent[22:32:29] BluetoothAPI: 0x7fe013d695b0
> ```

Did you set the log level to `Debug`?

vanessahsw, 2021-05-12:
I tried on another mac with updated OS and it was able to run. I will close the issue. Thanks for your help!

closed 2021-05-12 by vanessahsw