pybricks/support #329: The hub took too long to respond. Restart the hub and try again.

ferhatozdmr, 2021-05-09:
**Question**
I ghis error when I am trying to connect my movehub to macos 10.15.7

**Context**
What are you trying to accomplish that led to this question?

**Screenshots**
There is a saying that a picture is worth a 1000 words. Screenshots really help to identify and solve problems.

ferhatozdmr, 2021-05-09:
I have tried too many times. reinstalled original firmware and tried again. but I get the same error.

dlech, 2021-05-09:
Could you please try some of the things in our [Bluetooth Troubleshooting Guide](https://github.com/pybricks/support/discussions/270) to help us figure out what the problem is?

Which hub are you using?

ferhatozdmr, 2021-05-09:
Hi David, I couldn't get much info I guess. But here are some screenshots:
[image]
[image]

I can control it with out any problems through https://legoboost.azurewebsites.net/ this web site. I know it uses totally different protocol but I believe this shows that there is no hardware or communication issue between the hub and my computer. In pybricks case, it finds my hub, starts flashing the drive only at the last step it gives this error.

ferhatozdmr, 2021-05-10:
I have installed the firmware on another windows machine, and now I connect it to my mac without ant issues. I don't know what the problem is installing firmware on a mac but if anybody has similar problem on mac, install the firmware on a windows pc and use your hub with your mac.

dlech, 2021-05-10:
If you are able to capture the Bluetooth packets on your Mac, it would be interesting to compare it to #256.

dlech, 2021-05-17:
We just released `pybricks-code@1.0.0-rc.2` that fixed #256. Could you try flashing the firmware on your Mac using this new version to see if the problem on Mac is fixed as well?

ferhatozdmr, 2021-05-17:
> We just released ***@***.*** that fixed #256. Could you try flashing the firmware on your Mac using this new version to see if the problem on Mac is fixed as well?

Hi David,

Unfortunately still the same problem. I get the same error.”The hub took to long to respond” as soon as the installation reaches to %100. 

I did not have any problem when I try to flash using windows pc though.

dlech, 2021-05-17:
Thanks for checking. If you have time, it would still be helpful to do a Bluetooth packet capture on the Mac. It would be nice if we could fix it in case other people have the same problem.

ferhatozdmr, 2021-05-17:
Hi David, 

I will try to do my best for packet capture but I am not familiar how exactly to do it. I tried to follow your documentation but when I run the capture it seems like it is capturing everything, I couldn’t seperate just the bluetooth that the hub is connected to. Or I am doing it completely wrong :) I will give it another try.

dlech, 2021-05-17:
It's been a while since I did that so I don't remember exactly where it is, but there are some filters in the PacketLogger.app that can be enabled to narrow it down to just Bluetooth traffic.

ferhatozdmr, 2021-06-03:
Hi David,

Is there any plans for enabling urandom for Movehub? or is there any other
alternative that I can use randint for movehub? I tried to search all the
documentation but I couldn't find any answer.

dlech, 2021-06-03:
I posted about this recently: https://github.com/pybricks/support/discussions/319

ferhatozdmr, 2021-06-03:
Oh sorry, I didn't see it :) . Thank you for your quick response David.

ferhatozdmr, 2021-06-03:
One more question, is there a way to use input function in pybricks? I am
trying to get an input from a user to make the robot move different ways.

dlech, 2021-06-03:
Yes, there are several issues and discussions covering this topic.

ferhatozdmr, 2021-06-03:
Thanks a lot, I'll check it out.

laurensvalk, 2021-06-09:
> I get the same error.”The hub took to long to respond” as soon as the installation reaches to %100.

I think I can reproduce this consistently on Android with the City Hub.

And I can add/confirm that the hub then goes back to blinking pink, so there's no firmware on the hub; it's not just a matter of restarting the hub.

If this is not reproducable for you @dlech, I'll try to find logs as needed.

dlech, 2021-06-09:
I would rather start a new issue about Android since this one is about macOS.

laurensvalk, 2021-06-09:
It was more to confirm that it can in fact get to 100% and still be without a valid firmware, which we weren't sure about yet in a recent discussion.

laurensvalk, 2022-11-02:
On the Move Hub specifically, fresh batteries seem to make a big difference as well.

dlech, 2023-03-20:
Closing since Pybricks Code and the firmware has changed significantly since the initial report, so any reports of the same error message are not necessarily the same issue.

If anyone who has a problem that is fixed by turning off other Bluetooth devices, please capture Bluetooth logs as described [here](https://bleak.readthedocs.io/en/latest/troubleshooting.html#macos) and post the results in #938. This seems to b a common problem and could help other users if we can understand it better.

closed 2023-03-20 by dlech