pybricks/support #338: [Feature] Improve warning text for no adapter found

laurensvalk, 2021-05-17:
**Is your feature request related to a problem? Please describe.**
Sometimes, I forget to plug in the Bluetooth dongle and I get the expected error. But pluging it in does not resolve it. You have to restart the browser also. We could tell the user about this.

**Describe the solution you'd like**
Adjust the text to something like:

*No Bluetooth adapter could be found. Please insert a compatible adapter and restart the browser.*

[image]

dlech, 2021-05-17:
Should we also mention that it requires Bluetooth Low Energy?

laurensvalk, 2021-05-17:
Yes. Or -- I don't remember -- did we already have a "more info" link option for these notification? Maybe we could add a page to the site and link to it.

dlech, 2021-05-17:
It shouldn't be to hard to add a button. Just need the page first. :smile:

dlech, 2021-05-17:
We should also file a bug with Chromium. Restarting the browser should not be necessary.

dlech, 2021-06-30:
Looks like we did this already.

closed 2021-06-30 by dlech