pybricks/support #339: Warning message "No bluetooth" on connect pop-up overlays message about "no compatible device found"

BertLindeman, 2021-05-17:
**Describe the bug**
The two overlayed messages are illegible.
Seen this very minor bug on Windows and on Linux.

**To reproduce**
Steps to reproduce the behavior:
1. Go to `code.pybricks.com`
2. Switch off Bluetooth on your PC or whatever you use to connect to the hub
3. Click on the button `connect Bluetooth`
4. See the pop-up and try to read :wink:

**Expected behavior**
Either remove the message about "no compatible device" that is obvious if no Bluetooth is available,
or place the messages on their own line on the pop-up.

**Screenshots**
[image]

laurensvalk, 2021-05-17:
I've seen this a few times too. This may be a bug in Chromium. 

Thanks for providing a reproducible case!

BertLindeman, 2021-05-17:
Ah, an incomplete error report by me then :wink: 
On Windows seen on ```Microsoft Edge Versie 90.0.818.56 (Officiële build) (64-bits)```

laurensvalk, 2021-05-17:
Both Google Chrome and Microsoft Edge are built on Chromium :wink:

BertLindeman, 2021-05-17:
Indeed Laurens: on Win10 with ```Opera Versie:76.0.4017.107``` this scenario gets a red pop-up message:

[image]

dlech, 2021-05-17:
Here is a helpful link: https://www.chromium.org/developers/how-tos/file-web-bluetooth-bugs

Although in this case, I don't think any logs are necessary when filing the bug.

dlech, 2022-06-03:
Is this still a bug in chromium?

BertLindeman, 2022-06-03:
Tested (Now on WIN11) on beta.pybricks.com 
   with MSedge - no problem
   with chrome Version 102.0.5005.63 - also no problem.

closed 2022-06-03 by BertLindeman