pybricks/support #342: pybricksdev on Raspberry Pi

dlech, 2021-05-19:
**Is your feature request related to a problem? Please describe.**
I wanted to use `pybricksdev` on my headless Raspberry Pi to try to reproduce https://github.com/pybricks/support/discussions/334#discussioncomment-755174.

I ran into a couple of issues.

1. `pybricksdev` requires Python 3.8. The default Python on Raspberry Pi is currently 3.7. So I had to compile 3.8 from source. Not too painful if you have done it before but it does take some time.
2. There is not a `mpy-cross==1.12` binary wheel for Arm architectures. This should be fixed by https://github.com/pybricks/pybricksdev/commit/7a8a712a4de702b63253462cdcc5b8140dd64b8c.

**Describe the solution you'd like**
`pybricksdev` should be easy to install on Raspberry Pi, which means bumping the required Python version down to 3.7.

**Additional Context**

I don't remember why we set the version at 3.8. There could be APIs from the standard library we are using that were introduced in 3.8.

Debian bookworm is in freeze right now, so a new version of Debian should be coming some time this year but it could take a while longer for it to become the default on Raspberry Pi.

dlech, 2021-05-19:
After some more research, I have realized that Raspberry Pi currently has BluzZ 5.50 (Debian Buster). Since Bleak v0.11, this is broken. (As a side note, Pybricks Code is also broken on Raspberry Pi for a similar reason.)

So it probably isn't worth trying to support anything Pybricks on Raspberry Pi Buster. But we can support it when Raspberry Pi starts shipping Bullseye as the default. And dropping the minimum Python version in `pybricksdev` probably isn't worth it.

scotus-49, 2023-01-16:
I have been waiting for a long time to be able to use Pybricks with my Raspberry Pi. I am happy to say that it now seems to work. I read your comment above regarding Debian bullseye, and since the Raspi OS is now based on Debian version: 11 (bullseye), I thought it might now be at least worth a try.

Using a Raspi 4 with a standard install of the latest (64 bit) OS I had no problem installing Pybricks firmware on my Inventor Hub using Chrome browser. I then ran a couple of small test programs in Pybricks Code,(hub light and motor) and all worked as expected.

Running `bluetoothd -v` on my installed system gives `5.5`. I remember previously it was 5.0 and you had explained why it wouldn't work. I've been away from Pybricks for a while, and now look forward to exploring your new features with my RPi.

Thank you both for all your hard work.

Regards from Berlin.

Francis Sim

dlech, 2023-01-17:
Glad to hear it is working now.

closed 2023-01-17 by dlech