pybricks/support #343: Add motor calibration feature.

dlech, 2021-05-20:
**Is your feature request related to a problem? Please describe.**
Some LEGO Powered Up motors need to have the zero position calibrated. The LEGO MINDSTORMS app can do this, but other apps for other hubs don't have this feature yet.

**Describe the solution you'd like**
It would be nice if we had a script that we could copy and paste into Pybricks Code to to the calibration.

**Describe alternatives you've considered**
For people with a MINDSTORMS or SPIKE hub, they can just use the MINDSTORMS app.

**Additional context**
Related thread: https://github.com/pybricks/support/discussions/334#discussioncomment-760049

JorgePe, 2021-05-20:
` For people with a MINDSTORMS or SPIKE hub, they can just use the MINDSTORMS app.`

Currently, the MINDSTORMS App does not recognize Large Angular Motors (either SPIKE white/blue or Technic grey) as motors that support calibration.

dlech, 2021-05-21:
Here is a capture of the calibration procedure from the MINDSTORMS software. The motor is "reset" by stopping UART comms, then it is allowed to come up at 2400 baud rather than doing the initial 115200 baud setting. Then the baud rate is changed to 4800 baud. After the last write here, the hub switches back to 115200 baud but it takes the motor a while to reset again.

The attached data just contains the messages during the 4800 baud portion. I haven't really looked at it yet, so if someone wants to have some fun decoding the write messages, go for it.

[motor-calibration.csv](https://github.com/pybricks/support/files/6519476/motor-calibration.csv)

dlech, 2021-05-21:
It looks like it goes something like this:

- This repeats 3 times:
    - Set mode to 4 (CALIB)
    - Write 0x00
    - Write 0x03 + "LEGO-FAC-CAL-"
    - Wait 10 ms
    - Write 0x00
    - Wait 32 ms
    - **Write 0x02 0x00 0x40 0x00**
    - Wait 7 ms
    - Series of DATA commands
      - On first and third iterations, **hub sends NACK and receives DATA 20 times**
      - On second iteration, **hub sends DATA and receives ACK 5 times**
    - Write 0x01
    - Write 0x04
    - Set mode  to 0 (POWER) [default mode]
    - Wait about 1 second

- Then:
  - Set mode back to 4 (CALIB)
  - Write 0x03 + "LEGO-FAC-CAL-"
  - Wait 10 ms
  - Write 0x00
  - Wait 32 ms
  - **Write 0x02 0x40 0x41 0x00**
  - **Receive ACK**
  - Write 0x01
  - Write 0x04
  - Set mode  to 0 (POWER) [default mode]

- Then:
  - Set mode back to 4 (CALIB)
  - Write 0x03 + "LEGO-FAC-CAL-"
  - Wait 10 ms
  - Write 0x00
  - Wait 32 ms
  - **Write 0x02 0x48 0x41 0x00**
  - Wait 7 ms
  - **Hub sends 32-byte DATA and receives ACK 8 times**
  - Write 0x01
  - Write 0x04
  - Set mode  to 0 (POWER) [default mode]

- Then basically the same as the 1st iteration
  - Set mode back to 4 (CALIB)
  - Write 0x03 + "LEGO-FAC-CAL-"
  - Wait 10 ms
  - Write 0x00
  - Wait 32 ms
  - **Write 0x02 0x00 0x40 0x00**
  - **Hub sends NACK and receives 32-byte DATA 20 times**
  - Write 0x01
  - Write 0x04
  - Set mode  to 0 (POWER) [default mode]

Finally, stop sending NACKs to allow motor to resync at higher baud rate.

dlech, 2021-05-21:
So there appear to be 5 different "commands"
- 0x00: not sure what this does - sent once before 0x03 command and always before 0x02 command
- 0x01: not sure on this either - always sent before 0x04 command
- 0x02: I'm guessing the payload is a 16-bit address.
- 0x03: This is probably a "magic" value to unlock programming the flash
- 0x04: Seems to end the programming session

Reading and writing DATA probably reads/writes starting at the address specified in the 0x02 command.

dlech, 2021-05-21:
Same CSV with additional columns for DATA mesages:

[motor-calibration.csv](https://github.com/pybricks/support/files/6523873/motor-calibration.csv)