pybricks/support #347: [Bug] SystemExit and KeyboardInterrupt appear swapped on REPL

laurensvalk, 2021-05-26:
**Describe the bug**
- Pressing CTRL+C on the first line of the REPL exits the REPL (but it shouldn't).
- Running `raise SystemExit` does not exit the REPL (but it should).

**To reproduce**
Connect to the hub and start the REPL with the `>>>` button in Pybricks Code.

Click on the terminal pane and press CTRL+C. The run animation on the hub stops. If you run a few other commands first or wait a while, then CTRL+C does keep you in the REPL as expected.

For the second issue, start REPL, paste and execute `raise SystemExit`, and observe that it keeps running.

Tested on PrimeHub.

**Expected behavior**
- CTRL+C to just go to a new empty line on the REPL
- SystemExit to exit as if pressing the stop button.

**Screenshots**

CTRL+C on the first line triggers an exit:

[image]

If you don't do it right away, it does not:

[image]

laurensvalk, 2021-05-26:
When running scripts everything appears to be working as expected.

dlech, 2021-06-03:
>   * Running `raise SystemExit` does not exit the REPL (but it should).

AFAIK, not exiting the REPL on plain `SystemExit` is standard MicroPython behavior. There is an internal `PYEXEC_FORCED_EXIT` flag that is only set when `machine.soft_reset()` is called that modifies the behavior of `SystemExit`. (We also set `PYEXEC_FORCED_EXIT` with our stop buttons.)

dlech, 2021-06-03:
>  * Pressing CTRL+C on the first line of the REPL exits the REPL (but it shouldn't).

This is fixed now.

laurensvalk, 2021-06-03:
> AFAIK, not exiting the REPL on plain `SystemExit` is standard MicroPython behavior. There is an internal `PYEXEC_FORCED_EXIT` flag that is only set when `machine.soft_reset()` is called that [...]

OK - I suppose it doesn't exit in MicroPython because usually there is nothing to exit to. I guess I was extrapolating from how it works in scripts. It seems nice to make it the same. This isn't critical to me though.

dlech, 2021-06-03:
The change would need to be made in upstream MicroPython.

laurensvalk, 2021-06-03:
Then let's keep it simple and not change it :)

The CTRL+C was the main bug here, and it looks like you fixed it already. The system exit is just something that seemed odd when I dug deeper, but it's not really an issue.

closed 2021-06-03 by laurensvalk