pybricks/support #354: [Bug] Wrong line number reporting of (red) Syntax error in code editor

hoppy53, 2021-05-29:
Symptom:
if there is a syntax error caused in the middle of a line, the editor reports the correct line number where the error occurred. see attachment: code_editor_syntax_error_within_line.png
Here, as an example in line 10, the left parenthesis in   Motor Port.A)  is missing
Line number is correctly shown.

if there is a syntax error caused by the last character of a line, the editor reports instead the error being in the next line (line+1). see attachment: code_editor_syntax_error_end_of_line.png
Here, as an example in line 10, the right parenthesis in  Motor (Port.A  is missing.
Line number is not correct. It shows 11 instead of 10

It goes for all types of instructions as long as it is the last character of the line.
Also, it applies to both the webpage editor and the chrome app install.

[image]
[image]

Environment: 
Linux Mint 20.1 Ulyssa
Chrome Version 91.0.4472.77

dlech, 2021-05-29:
In Python and MicroPython, it is allowable to continue on the next line after an `(` in the code. So the compiler doesn't realize that `)` is missing until something else unexpected is reached in the code.

As a side note, I actually read somewhere recently that CPython got a compiler upgrade that will actually report the error on the line with the `(` as expected. But since we are using MicroPython, I don't expect it to get the same fix. In MicroPython, often keeping the firmware size to a minimum is more important than niceties like this.

dlech, 2021-05-29:
Although, we may be able to add other checkers to Pybricks code in the future that perform additional analysis. The trick is being able to run Python in the web browser or find JavaScript/TypeScript tools for analyzing Python code.

davecparker, 2022-12-30:
> Although, we may be able to add other checkers to Pybricks code in the future that perform additional analysis. 

Yes, and here are two more ideas for you to kick around:

1. As an alternative to finding web-ready analysis code, you might also look for a free server-based solution you could use. I wouldn't be surprised if some university project did one, perhaps even using AI. Or maybe you could even put a Python 3.10 parser (has improved syntax errors) on your own server and call it yourself.

2. Consider analyzing some simple situations yourself. I have done similar things with Java. For example, if the parser tells you the syntax error is on line n, try re-parsing with line n replaced by "pass" and again with "(True)". If both of these also fail at line n at the first column (hopefully the parser tells you the column internally), then pretty high confidence that the error is actually on line n-1. So... if the first token of line n-1 is "if", "elif", "else", "while", "for", or "def" and the last token of line n-1 is not ":" then guess what :)   Not 100% confidence, so word the error as "missing colon?" , but it almost always will be, and if you're wrong the user is not in much worse of a situation than they were before.

davecparker, 2022-12-30:
> 2. Consider analyzing some simple situations yourself. 

I should add that in the case of missing : you can test the hypothesis by adding a : and re-parsing and see if it passes. I have done this to great success with missing ; against a weak Java parser.