pybricks/support #357: Returning generator object for non-blocking code.

dlech, 2021-05-31:
Using generators as coroutines is a nice way to create non-blocking code in MicroPython (see https://github.com/pybricks/support/discussions/356).

Currently, for motors we have to write something like this to get non-blocking, cancellable motor operations:

```python
def example():
    motor.run_angle(400, 90, wait=False)
    try:
        while not motor.control.done():
            yield
    catch GeneratorExit:
        motor.stop()
```

However, if we modify the `run_angle()` method to return a generator object, we can simply write:

```python
def example():
    yield from motor.run_angle(400, 90, wait=False)
```

I made a couple of proof-of-concept implementations:
- https://github.com/pybricks/pybricks-micropython/commit/b46c98c1522e2edf1a04b48c98074fbbd51c16ae
- https://github.com/pybricks/pybricks-micropython/commit/248710a0ad18e8d4c00abc86f1c87bd924c94ffb

The first one takes about 200 bytes more firmware size than the second. Also the second one doesn't allocate any new memory when used.

Thinking about these brings up some interesting cases.
1. What should happen when a motor is stalled and the operation does not complete?
2. Should it be allowed for other tasks to use the same motor? Usually starting a new motor command will cancel one that is running in the background. However the current method of waiting on the control done flag won't work since there is no way to differentiate between the old command and the new command. A generator waiting on the old command will only see that the command is not done and keep waiting even though the command it started was interrupted.
3. The `run_until_stalled()` method returns a value, so we can't have a `wait=False` parameter there.
4. The methods `run()`, `dc()` and `track_target()` are implicitly `wait=False` so they should probably return the generator object as well so that they can be passed to composing functions and so that they can be cancelled.

dlech, 2021-05-31:
We will also need to go through the API and see what other blocking and background functions there are that would also make sense as generators. Off of the top of my head, there are light animations, sound playback (on SPIKE/MINDSTORMS), switching modes on sensors and writing data to sensors.

Having long running object initializes, on the other hand, should be permissible since those happen at the beginning of the program before the main control/event loop starts running.

dlech, 2021-06-01:
There are also some potentially blocking built-in functions like `print()` (related #305).

azzieg, 2022-10-04:
It must be possible, since the original firmware seems to have nailed it (my findings: https://github.com/azzieg/mindstorms-inventor/tree/main/word_blocks). From what I understand:
1. Original hub firmware provides control over stall detection. If stall detection is on, which is the default, and the motor is blocked, then the operation will complete early. If stall detection is off, if might never complete if the motor is stalled. When the operation completes, the status is saved. Possible outcomes are: success (target reached), interrupted (overridden by another command) or stalled (if stall detection is on).
2. Could we somehow cancel the old command when a new command is issued? If we had something like per-port operation counter, we could find out that our operation was overridden, I'd think.
3. If I understood correctly, generators can both yield and return a final value. The final value can be extracted from StopIteration's value.
4. The original firmware has a number of "start" operations which do not return generators. I guess it's because those are not supposed to be awaited and if the "async" operators (that return generators) are not awaited (the generator is not iterated in the event loop) the the motors don't even start.

laurensvalk, 2022-11-25:
> I made a couple of proof-of-concept implementations

This is brilliant! I didn't realize we could use this with `async`/`await` already without enabling additional MicroPython features that take up more size.

> The first one takes about 200 bytes more firmware size than the second. Also the second one doesn't allocate any new memory when used.

We might need something more like the first one depending on the points you raised in (2). 

> The methods run(), dc() and track_target() are implicitly wait=False so they should probably return the generator object as well so that they can be passed to composing functions and so that they can be cancelled.

These methods should't need one. To "cancel" them you can just call `.stop()` or any other new async motor command to take over.

> The run_until_stalled() method returns a value, so we can't have a wait=False parameter there.

This can be addressed with the approach below

> What should happen when a motor is stalled and the operation does not complete?

The same as it does today. (But parallel async functions just keep running).

**Async API**

Instead of relying on partially existing `wait` parameters on some methods, we could just add one parameter when initializing:

```python
# All motor operations will be async
my_motor = Motor(Port.A, use_async=True)
```

```python
# All motor operations will block.
my_motor = Motor(Port.A)
```

This has a few benefits:
- No braking change to API
- Don't need to add `wait=False` everywhere. Having `await something(wait=False)` gets confusing.
- Solves `run_until_stalled` even though it doesn't have `wait` parameter
- This solves sensors too. If initialized with `use_async=True`, all read methods can be async. This solves waiting on mode switches.

This also provides a really neat balance between simplicity (beginning users can still use blocking code) and functionality. 

If this gets more widely adopted, we can quietly deprecate the `wait` parameter from the docs. We could still leave it implemented for backwards compatibility until perhaps we arrive at Pybricks 4.0.

laurensvalk, 2022-11-25:
> Should it be allowed for other tasks to use the same motor? Usually starting a new motor command will cancel one that is running in the background. However the current method of waiting on the [...]

This is where your first approach might come in handy, where there's one generator object per command.

Instead of polling the motor status, we could register a callback with the motor to be called on completion/interruption/stall.

The motor code is pretty much ready for this. However we need to make sure those callbacks are unregistered when the program ends so the motor drivers won't call them back after MicroPython is done.

We might already be a bit closer now we've cleaned up the whole download and run. If we get this right, we'll also be able to dynamically allocated things like DriveBase structures only when they are used. This will get us back some of the RAM we'll use for the generators :)

laurensvalk, 2023-01-10:
> I made a couple of proof-of-concept implementations:

There is a nice middle-ground here. The motor object can have one static `pb_type_MotorWait_t` object singleton. This can be used for a new motor command if the motor isn't in use already. Only if it is in use, allocate another.

This avoids allocating on every motor command in "good" programs, but we could make all programs work correctly by allocating if needed.

laurensvalk, 2023-01-12:
> I made a couple of proof-of-concept implementations:
> - https://github.com/pybricks/pybricks-micropython/commit/b46c98c1522e2edf1a04b48c98074fbbd51c16ae
> - https://github.com/pybricks/pybricks-micropython/commit/248710a0ad18e8d4c00abc86f1c87bd924c94ffb

I expanded the first approach to address some of the design questions in OP in https://github.com/pybricks/pybricks-micropython/compare/1e3447c69a0c434d725e5e8bc6c52d2f529e5319..1e5820c4eb60cf3bfd9f85a8b4b7fe941fc2057f:

- Avoids allocation on each command.
   - This avoids excessive allocations in loops.
   - Only allocates N times if N tasks use the motor at once. N is small in practice, or 1 in a "good" script.
- If another task starts a motor command, all other tasks for this motors are cancelled.
    - This avoids other tasks waiting possibly indefinitely until the motor is "done".
    - Instead, the task just proceeds.
    - We could choose to raise an exception, or pass gracefully as we do already.
- Implements `await` for `run_until_stalled` as well
    - Some more work may be needed if we want to restore the duty cycle limit on exceptions.

Code example:

```python
from pybricks.hubs import PrimeHub
from pybricks.pupdevices import Motor
from pybricks.parameters import Port, Direction, Color
from itertools import zip_longest as run_parallel

# Use async mode for this script. Revisit names for this API.
from pybricks import set_async_mode
set_async_mode(True)

# Initialize motors as usual.
left = Motor(Port.A, Direction.COUNTERCLOCKWISE)
right = Motor(Port.B)
hub = PrimeHub()

# Motor operations can be awaited:
async def example():
    print("Spin left motor!")
    await left.run_angle(500, 4 * 360)
    print("Spin left motor again!")
    await left.run_angle(500, -360)

# So, this code can run in parallel:
async def blink():
    for i in range(7):
        hub.light.on(Color.RED)
        await right.run_time(500, 500)
        hub.light.on(Color.GREEN)
        await right.run_time(-500, 500)

# Run both tasks in parallel. # Revisit: Make lightweight generic API for this.
for _ in run_parallel(example(), blink()):
    pass
```

Note, running this code requires having a separate file in Pybricks code called `itertools.py` with the following contents:

```python
def repeat(object, times=None):
    # repeat(10, 3) --> 10 10 10
    if times is None:
        while True:
            yield object
    else:
        for i in range(times):
            yield object

def zip_longest(*args, fillvalue=None):
    # zip_longest('ABCD', 'xy', fillvalue='-') --> Ax By C- D-
    iterators = [iter(it) for it in args]
    num_active = len(iterators)
    if not num_active:
        return
    while True:
        values = []
        for i, it in enumerate(iterators):
            try:
                value = next(it)
            except StopIteration:
                num_active -= 1
                if not num_active:
                    return
                iterators[i] = repeat(fillvalue)
                value = fillvalue
            values.append(value)
        yield tuple(values)
```

laurensvalk, 2023-01-12:
The main logic here is for the purpose of cancellation. 

After that, not allocating is really just a few lines of code.

laurensvalk, 2023-03-09:
This is being continued in 917.

closed 2023-03-09 by laurensvalk