pybricks/support #358: Include select functions from itertools

dlech, 2021-05-31:
I like using the [itertools](https://docs.python.org/3/library/itertools.html) library. MicroPython doesn't have this in the core. There are a few functions defined in [micropython-lib](https://github.com/micropython/micropython-lib).

On hubs with enough flash space, it would be nice to include the most useful functions as a frozen module.

The ones I use the most are [count](https://docs.python.org/3/library/itertools.html#itertools.count) and [chain.from_iterable](https://docs.python.org/3/library/itertools.html#itertools.chain.from_iterable).

Also useful to me: [cycle](https://docs.python.org/3/library/itertools.html#itertools.cycle), [repeat](https://docs.python.org/3/library/itertools.html#itertools.repeat), [zip_longest](https://docs.python.org/3/library/itertools.html#itertools.zip_longest).

laurensvalk, 2021-06-07:
I've begun to appreciate [chain](https://docs.python.org/3/library/itertools.html#itertools.chain) as well.

dlech, 2021-06-07:
I don't know if I would use it in Pybricks that much, but in regular Python it seems like I usually need it instead of `chain()`.

laurensvalk, 2021-07-28:
With the way we organize the docs and compatibility now, this could integrate quite nicely.

If we did this, would the user import it from `itertools`, `uitertools`, or somewhere else entirely?

It seems like there's ways to argue for both, so I'm not sure which to pick yet.

dlech, 2021-07-28:
If we are doing standard-library compliant functions, then `uitertools` would make sense. If we want to do [something specifically for use with generator functions](https://github.com/pybricks/support/discussions/356) instead, then `pybricks.tools` might make more sense.

laurensvalk, 2022-08-26:
Generalizing this, we could add modules like these to Pybricks Code. Either invisibly (and just document them), or as read-only, so people can see what's inside.

Then when a user imports (from) such as module, we could include it using the idea in https://github.com/pybricks/pybricks-micropython/pull/115

laurensvalk, 2022-11-28:
Idea for future releases:

We can essentially support this now by shipping `itertools` when it is used. But for most cases this will include many unused functions too.

Maybe there are Python modules than can strip any non-imported (and unreferenced) code from a module.

Or, if itertools is the only special case that needs this in practice, just generating an "intertools" module on the fly with only the ones used could be feasible.

laurensvalk, 2024-10-02:
These can now be included as a module that you can import, so I think we can close this.

Although we could also include these as a frozen Python module now. Feel free to re-open if you think we should do this.

closed 2024-10-02 by laurensvalk