pybricks/support #362: Robot Inventor / SPIKE Prime beta installer problems

laurensvalk, 2021-06-03:
> Hi @laurensvalk,
> 
> I've just connected the Mindstorms Inventor Hub to the laptop through USB, but https://beta.pybricks.com/ shows "No compatible devices found". My screenshot is attached.
> 
> Should I have done some step differently?
>
> Thank you @laurensvalk.
>
> Vinh

[image]

_Originally posted by @TheVinhLuong102 in https://github.com/pybricks/support/issues/186#issuecomment-853646081_

laurensvalk, 2021-06-03:
Did this work before? How did you install Pybricks previously?

Please make sure that no other apps are currently using the hub.

TheVinhLuong102, 2021-06-03:
Hi @laurensvalk,

> Did this work before? How did you install Pybricks previously?

Previously, I installed PyBricks on the Inventor & Prime hubs using the .llsp & .lms scripts runnable through the official apps.

> Please make sure that no other apps are currently using the hub.

No, all LEGO apps were turned off, so no other programs were using the hub.

Vinh

TheVinhLuong102, 2021-06-03:
This is my first time using https://beta.pybricks.com.

laurensvalk, 2021-06-03:
Could you try it with this turned on?

[image]

TheVinhLuong102, 2021-06-03:
@laurensvalk,

I've turned that on (see attached) and restarted the browser, but it still doesn't solve the problem.

[image]

laurensvalk, 2021-06-03:
> Btw, I've tried using 3 different laptops, and this doesn't work on any of them.

That's really interesting, as nobody has reported this yet. Which OS'es are these?

Was USB plugged in in all cases?

TheVinhLuong102, 2021-06-03:
I'm using Mac, with OSX Big Sur.

And yes, USB connection worked fine with the official Mindstorms app. We just upgraded the official firmware earlier this evening.

laurensvalk, 2021-06-03:
Thanks for sharing, I guess we'll need to look into this.

Although not related to Pybricks, does this [generic web terminal work for you](https://bipes.net.br/aroca/web-serial-terminal/) if you select `115200` for the speed?

TheVinhLuong102, 2021-06-03:
Oh @laurensvalk, I've just realized I used an *unsupported USB cable*, which worked fine for charging the battery but not for connecting the Hub through USB.

I've just grabbed the official USB cable included with the Inventor/Prime sets and it seems the installation is progressing.

TheVinhLuong102, 2021-06-03:
@laurensvalk, however, the installation failed with the following messages:

Installation progress
Something went wrong. Please try again.
from _pybricks.install_pybricks import install; install()
Starting installation script.
Checking installation files.
The installation files are corrupt.
>>>

TheVinhLuong102, 2021-06-03:
When I refreshed that page to redo the installation, then the progress seemed to hang at the following stage:

Installation progress
Uploading firmware-dual-boot-base.bin: 100%

TheVinhLuong102, 2021-06-03:
After about 15 more minutes, I turned off the Hub, turned it on again, and refreshed the Beta installer page and tried to redo. The progress then again seemed to hang at the following stage:

Installation progress
Uploading firmware-dual-boot-base.bin: 58%

TheVinhLuong102, 2021-06-03:
Another retry, now with the following error messages:

Installation progress
Something went wrong. Please try again.
from _pybricks.install_pybricks import install; install()
Starting installation script.
Checking installation files.
The installation files are corrupt.
>>> {"i":"-9Uf","m":"get_hub_info","p":{}}
{'m': 'get_hub_info', 'i': '-9Uf', 'p': {}}
>>> {"i":"boEv","m":"get_hub_info","p":{}}
{'m': 'get_hub_info', 'i': 'boEv', 'p': {}}
>>> {"i":"HHdm","m":"get_hub_info","p":{}}
{'m': 'get_hub_info', 'i': 'HHdm', 'p': {}}
>>> {"i":"nsP4","m":"get_hub_info","p":{}}
{'m': 'get_hub_info', 'i': 'nsP4', 'p': {}}
>>> {"i":"KqES","m":"get_hub_info","p":{}}
{'m': 'get_hub_info', 'i': 'KqES', 'p': {}}
>>> {"i":"l0ZJ","m":"get_hub_info","p":{}}
{'m': 'get_hub_info', 'i': 'l0ZJ', 'p': {}}
>>> {"i":"OMAu","m":"get_hub_info","p":{}}
{'m': 'get_hub_info', 'i': 'OMAu', 'p': {}}
>>>

TheVinhLuong102, 2021-06-03:
My latest attempt:

- Reset the Hub totally using the Mindstorms official app: wiped out & installed the official firmware again.

- Used beta.pybricks.com installer.

- The progress seemed to hang at the following stage: Uploading firmware-dual-boot-base.bin: 100%. The 5x5 light matrix turned on for 10 minutes, then the whole light matrix turned off. The Hub didn't restart itself.

When I manually boot up the Hub and tried to turn on PyBricks by holding the button long, it didn't bring me to PyBricks but instead the usual official interface. https://code.pybricks.com/ couldn't detect the Hub, either.

laurensvalk, 2021-06-03:
Thanks for trying this. Glad the cable got us a step further. I’ll look at the other issues later tonight. Hope you don’t mind the wait :)

TheVinhLuong102, 2021-06-03:
@laurensvalk sure, thank you so much! :)

laurensvalk, 2021-06-03:
Which version of the mindstorms app did you use to install the official firmware?

TheVinhLuong102, 2021-06-03:
@laurensvalk it's Mindstorms Inventor app Version 10.1.0 (860509)

TheVinhLuong102, 2021-06-03:
My latest error messages are as follows:

Installation progress
Running the installation script.
from _pybricks.install_pybricks import install; install()
Starting installation script.
Checking installation files.
Files looking good!
Getting firmware info.
Your hub runs LEGO Firmware version: b'v1.2.01.0103-d08b6fe'
Initializing flash for 981240 bytes.
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "_pybricks/install_pybricks.py", line 195, in install
KeyboardInterrupt: 
>>> {"i":"aLLf","m":"get_hub_info","p":{}}
{'m': 'get_hub_info', 'i': 'aLLf', 'p': {}}
>>> {"i":"xGxk","m":"get_hub_info","p":{}}
{'m': 'get_hub_info', 'i': 'xGxk', 'p': {}}
>>> {"i":"RXdk","m":"get_hub_info","p":{}}
{'m': 'get_hub_info', 'i': 'RXdk', 'p': {}}
>>> {"i":"_8aQ","m":"get_hub_info","p":{}}
{'m': 'get_hub_info', 'i': '_8aQ', 'p': {}}
>>> {"i":"y2yI","m":"get_hub_info","p":{}}
{'m': 'get_hub_info', 'i': 'y2yI', 'p': {}}
>>> {"i":"uMnD","m":"get_hub_info","p":{}}
{'m': 'get_hub_info', 'i': 'uMnD', 'p': {}}
>>> {"i":"5FkV","m":"get_hub_info","p":{}}
{'m': 'get_hub_info', 'i': '5FkV', 'p': {}}
>>> {"i":"lvJt","m":"get_hub_info","p":{}}
{'m': 'get_hub_info', 'i': 'lvJt', 'p': {}}
>>> {"i":"aoDR","m":"get_hub_info","p":{}}
{'m': 'get_hub_info', 'i': 'aoDR', 'p': {}}
>>> {"i":"KDMP","m":"get_hub_info","p":{}}
{'m': 'get_hub_info', 'i': 'KDMP', 'p': {}}
>>> {"i":"DXQN","m":"get_hub_info","p":{}}
{'m': 'get_hub_info', 'i': 'DXQN', 'p': {}}
>>> {"i":"twkd","m":"get_hub_info","p":{}}
{'m': 'get_hub_info', 'i': 'twkd', 'p': {}}
>>> {"i":"jncc","m":"get_hub_info","p":{}}
{'m': 'get_hub_info', 'i': 'jncc', 'p': {}}
>>> {"i":"C-QK","m":"get_hub_info","p":{}}
{'m': 'get_hub_info', 'i': 'C-QK', 'p': {}}
>>> {"i":"pgh4","m":"get_hub_info","p":{}}
{'m': 'get_hub_info', 'i': 'pgh4', 'p': {}}
>>> {"i":"edLu","m":"get_hub_info","p":{}}
{'m': 'get_hub_info', 'i': 'edLu', 'p': {}}
>>> 
>>> 
>>> 
raw REPL; CTRL-B to exit
>OK
>raw REPL; CTRL-B to exit
>OK
>raw REPL; CTRL-B to exit
>OK
>raw REPL; CTRL-B to exit
>OK
>raw REPL; CTRL-B to exit
>OK
>raw REPL; CTRL-B to exit
>OK
>raw REPL; CTRL-B to exit
>OK
>raw REPL; CTRL-B to exit
>OK
>raw REPL; CTRL-B to exit
>OK
>raw REPL; CTRL-B to exit
>OK
>raw REPL; CTRL-B to exit
>OK
>raw REPL; CTRL-B to exit
>OK
>raw REPL; CTRL-B to exit
>OK
>raw REPL; CTRL-B to exit
>OK
>raw REPL; CTRL-B to exit
>OK
>OK>raw REPL; CTRL-B to exit
>OK
>OK>raw REPL; CTRL-B to exit
>OK
>OK>raw REPL; CTRL-B to exit
>OK
>OK>raw REPL; CTRL-B to exit
>OK
>OK>raw REPL; CTRL-B to exit
>OK
>OKv1.2.01.0103-d08b6fe
>raw REPL; CTRL-B to exit
>OK
>OKv1.2.01.0103-d08b6fe
>raw REPL; CTRL-B to exit
>OK
>OKv1.2.01.0103-d08b6fe
>raw REPL; CTRL-B to exit
>OK
>OKv1.2.01.0103-d08b6fe
>raw REPL; CTRL-B to exit
>OK
>OKv1.2.01.0103-d08b6fe
>raw REPL; CTRL-B to exit
>OK
>

laurensvalk, 2021-06-03:
To help us debug, are you able to access the hub REPL in the standard firmware via USB in a terminal application on your computer? Let me know if you need help with this. 

If you prefer, there's also the Thonny editor that can be quite nice.

Since the files appear to have come across correctly on your last try, it would be interesting to see the output of:

from _pybricks.install_pybricks import install; install()

TheVinhLuong102, 2021-06-03:
Absolutely @laurensvalk -- we're most happy to help with the debug!

Please let me know the devtools to install and how to get into the hub's FS or REPL.

TheVinhLuong102, 2021-06-03:
@laurensvalk, I've just tried installing another time, and this time it was a Success!

> Installation progress
> Done!
> Please wait for the hub to restart. Then start coding!

There's nothing I did differently this time. I just retried the same way, and it happened to work now.

TheVinhLuong102, 2021-06-03:
But then, after the Pybricks firmware installation succeeded, when I turned on the Pybricks mode (by holding the hub button long), the hub would immediately hang after the 3x3 light matrix appeared.

laurensvalk, 2021-06-03:
Let's leave it at that for now, then. We can debug it another time. Enjoy the latest firmware :)

laurensvalk, 2021-06-03:
What do you mean by hang? Does the blue light not blink? Does it not connect to pybricks code? 

There's no program by default, so that's why you'll see the 3x3 stop sign.

TheVinhLuong102, 2021-06-03:
@laurensvalk problem is the firmware is not usable yet. :|

The Hub keeps hanging. And when I take the battery out and put it in again after 30 seconds or so, the 3x3 PyBricks light matrix would return and the Hub is not responsive.

TheVinhLuong102, 2021-06-03:
@laurensvalk ok I've managed to boot Pybricks again now after taking out the battery and putting it in again. The Hub seems responsive now.

I'll proceed to testing this today :)

Thank you very much!

laurensvalk, 2021-06-03:
Can you try again with USB unplugged? Or was it unplugged already?

Removing the battery while USB is still plugged in doesn't really do anything because it still has power.

laurensvalk, 2021-06-03:
OK, sounds like you figured it out 🎉

To confirm, are you able to boot the standard firmware still as well?

TheVinhLuong102, 2021-06-03:
Yes, the standard firmware also boots up correctly. :)

TheVinhLuong102, 2021-06-03:
btw @laurensvalk, is this statement still true?

> Your hub must run the latest official MINDSTORMS firmware, even if you use a SPIKE Prime hub. Use the MINDSTORMS app (v1.3.4 [10.1.0] or newer) to install it.

I have a Prime Hub and can try the installation process on it too. Do I first have to wipe out the Spike Prime firmware and flash the Inventor firmware before installing Pybricks?

laurensvalk, 2021-06-03:
Thanks, but don’t worry too much about testing this installer for us at this stage. It’s really just an experimental tool at this stage, and certainly not very polished. We’ll be doing this much more thoroughly when we focus more on these hubs. 

And yes, the statement is still true for the time being. So, only do it if you really want to run it on that hub. 😄

TheVinhLuong102, 2021-06-03:
@laurensvalk I'll do it then -- converting the Prime hub to an Inventor hub and then installing the latest PyBricks on. That way, I can also help you test the installation process another time.

Antoni & Ariana really want to try out the Remote functionalities, and there'll be a noisy conflict if there's only 1 hub. :D

TheVinhLuong102, 2021-06-03:
So here's what I've done and seen with the installation process on the Prime Hub.

1ST ATTEMPT:

- I used Mindstorms app [Version 10.1.0 (860509)] to install the latest official HubOS [version 3.1.7.2] on the Prime hub, replacing the SPIKE Prime firmware

- I used https://beta.pybricks.com/ to install PyBricks

> The progress seemed to hang at the following stage: Uploading firmware-dual-boot-base.bin: 100%. The 5x5 light matrix turned on for 10 minutes, then the whole light matrix turned off. The Hub didn't restart itself.

> When I manually boot up the Hub and tried to turn on PyBricks by holding the button long, it didn't bring me to PyBricks but instead the usual official interface. https://code.pybricks.com/ couldn't detect the Hub, either.

The same things happened to the Prime Hub

TheVinhLuong102, 2021-06-03:
2ND ATTEMPT (on Prime hub with official Mindstorms firmware):

TheVinhLuong102, 2021-06-03:
3RD ATTEMPT (on Prime hub with official Mindstorms firmware): SUCCESS!!

> Installation progress
> Done!
> Please wait for the hub to restart. Then start coding!

So @laurensvalk we **can** indeed succeed, but after multiple tries.

laurensvalk, 2021-06-04:
Thanks! Yeah, other than the color this is the same hub, so this is more or less expected.

I suspect that something might be off with the timing causing it to fail sometimes. I can probably try to slow things down just slightly.

The reason for requiring the Inventor Hub firmware is that it comes with a fast file transfer feature. Perhaps the SPIKE firmware will get it too in the next app update. Then we can remove that note.

TheVinhLuong102, 2021-06-04:
Thank you @laurensvalk @dlech for the brilliant work!!

deopark, 2021-06-04:
on spike prime 1.3.5 firmware hub...
In my case, It is always doesn't restart hub(down... ) after install pybricks firmware in (https://beta.pybricks.com/)

laurensvalk, 2021-06-04:
Could you elaborate a little bit? You need the MINDSTORMS Robot Inventor firmware on the hub to try this.

deopark, 2021-06-05:
I received below image from (https://beta.pybricks.com/)

[image]

I can't turn on the hub after that.

laurensvalk, 2021-06-05:
Looks like you were using dfu.pybricks.com, not beta.pybricks.com

That’s fine though, but maybe you picked the wrong firmware file.

Try DFU again with this file to restore the original firmware:
https://github.com/gpdaniels/spike-prime/raw/master/firmware/storm%20-%20v1.2.01.0103-d08b6fe%20-%203.1.7.2/63c426a72bf3bb6fbdf6c3c32ac82859.bin

deopark, 2021-06-06:
thank you!! 
My hub woke up via (https://dfu.pybricks.com/)

I was confusing pybricks and official.

I don't know much about pybrics yet.
I will check it!!

deopark, 2021-06-06:
Thank You!

My hub woke up via (dfu.pybricks.come )

Pybricks seems difficult to me yet.
I'll check it more before I ask again.

TheVinhLuong102, 2021-07-22:
@laurensvalk I've tried install.pybricks.com again for our 2 hubs and the firmware installation went very smoothly this time. I think we can close this issue. :)

Thank you very much!

laurensvalk, 2021-07-27:
I've now improved this, and also enabled SPIKE Prime support (with SPIKE firmware).

If you have time, could you retry for both Robot Inventor and SPIKE Prime? That would be greatly appreciated.

TheVinhLuong102, 2021-07-30:
@laurensvalk we've just tried install.pybricks.com again with both:
- an Inventor Hub running the official MS firmware, and
- a Prime Hub running the official SPIKE Prime firmware
and the installation succeeded in both cases. 🎉 

I think this issue is totally resolved now. :)

Vinh

dlech, 2022-07-28:
We can now install firmware on SPIKE hubs in Pybricks Code v3.2.0b3 (Pybricks Beta v2.0.0-beta.3) so this issue is no longer relevant.

closed 2022-07-28 by dlech