pybricks/support #366: [Bug] Jerky motor movement, especially on low speeds ?

Vinz1911, 2021-06-06:
**Describe the bug**
I'm not sure if this is a "Bug" but I discovered a jerky motor movement on Technic Hub and on Spike/Inventor Hub. It looks Like that the motors are not linear turning, they frequently stutter. If I'm using Motor class only or drive base makes no difference. I discovered that also on higher speeds but on lower speeds you can see it much better. 

I testet the original lego firmwares to check if this is a normal behavior but with the original firmwares the problem does not exists. I made two videos, one with the lego firmware and one with the latest Pybricks Firmware. 

**To reproduce**
Steps to reproduce the behavior:
just create a motor instance in code and let them turn. 

**Expected behavior**
same fluently linear turning like the original firmware

**Screenshots**
I created two videos, one with original fw and one with pybricks fw. I hope you can see what I mean.

https://user-images.githubusercontent.com/21056653/120925423-dbee5f80-c6d8-11eb-9e82-a95ce75dd13a.mp4

https://user-images.githubusercontent.com/21056653/120925431-eb6da880-c6d8-11eb-8595-2d64c53b6656.mp4

laurensvalk, 2021-06-06:
Thanks for taking the time to investigate and report this!

I think we’ll have to improve the default settings a bit to make this better, especially for this motor type.

There are several settings that affect this, so I’ll have a look at trying to find the right balance, without making performance worse for other applications.

Vinz1911, 2021-06-06:
Thank you for your fast reply. If possible it would be very very nice to get a bit more smoother motor movement :) 

Thank you

laurensvalk, 2021-09-28:
This should be a bit better now. Would you mind trying this with the latest build?

[Technic Hub firmware](https://github.com/pybricks/pybricks-micropython/suites/3893404619/artifacts/97046890). [Other hubs](https://github.com/pybricks/pybricks-micropython/actions/runs/1281790816).

Vinz1911, 2021-09-28:
Hi Laurens, sounds great. I will try it now :)

Vinz1911, 2021-09-28:
Works great. The movement is much smoother now. 👍

laurensvalk, 2021-09-28:
Thanks for testing so quickly. I'll close this one then.

Feel free to open new issues for motor aspects when you run into them.

laurensvalk, 2023-03-06:
https://user-images.githubusercontent.com/12326241/223133084-09e43279-761a-44d3-b5ae-03fffc59d368.mp4

laurensvalk, 2023-03-06:
Some more work has been done here. Should be even better now, see above.

Vinz1911, 2023-03-06:
Hey @laurensvalk 👋🏻

I tried it and seems to work very well, great job 🥳